TillFleisch/ESPHome-Philips-Smart-Coffee #2: Having issues using esphome  2023.3.2 and example.yaml

schlunsen, 2023-03-29:
First of all - really nice project and thanks for publishing the code!

I think the current example.yaml isn't up to date with newest version of esphome.

What version of esphome are you currently using?

I'm trying to run `esphome run example.yaml`, but getting the following errors
```
display_uart: uart_display
  mainboard_uart: uart_mainboard
  power_pin: GPIO12
  id: philip
text_sensor.philips_status_sensor: [source example.yaml:43]

  Platform not found: 'text_sensor.philips_status_sensor'.
  platform: philips_status_sensor
  controller_id: philip
  name: Status
switch.philips_power_switch: [source example.yaml:48]

  Platform not found: 'switch.philips_power_switch'.
  platform: philips_power_switch
  controller_id: philip
  name: Power
  icon: mdi:coffee-maker
```

TillFleisch, 2023-03-29:
I updated to 2023.3.2, and my .yaml file compiled successfully.

After adding
```
esphome:
  name: example
  friendly_name: example
```
to the [example.yaml](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/blob/6602c047c5be34efb0a6ec42af0bb32b5c4a3d49/example.yaml), it also compiled successfully.

It looks like your configuration cannot find the custom component.
Make sure you add the component by adding the `external component` section to your yaml like this:
```
external_components:
  - source: github://TillFleisch/ESPHome-Philips-Smart-Coffee@main
```

If you are using a local copy of the repository, ensure the selected path is correct.

schlunsen, 2023-03-29:
Still having issues even when using github as external source.

Will look into it later this week, but I do see the components being downloaded to the .esphome/external_components

My example.yaml

```
esphome:
  name: example
  friendly_name: example

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  # Disable serial logging
  baud_rate: 0

external_components:
  - source: github://TillFleisch/ESPHome-Philips-Smart-Coffee@main

uart:
  # UART connected to the mainboard
  - tx_pin: GPIO1
    rx_pin: GPIO3
    baud_rate: 115200
    id: uart_mainboard

  # UART connected to the display
  - tx_pin: GPIO14
    rx_pin: GPIO13
    baud_rate: 115200
    id: uart_display

philips_series_2200:
  display_uart: uart_display
  mainboard_uart: uart_mainboard
  power_pin: GPIO12
  id: philip

text_sensor:
  - platform: philips_status_sensor
    controller_id: philip
    name: "Status"

switch:
  - platform: philips_power_switch
    controller_id: philip
    name: "Power"
    icon: mdi:coffee-maker
```

And still with the same error messages:

```
Failed config

philips_series_2200: [source example.yaml:33]

  Component not found: philips_series_2200.
  display_uart: uart_display
  mainboard_uart: uart_mainboard
  power_pin: GPIO12
  id: philip
text_sensor.philips_status_sensor: [source example.yaml:39]

  Platform not found: 'text_sensor.philips_status_sensor'.
  platform: philips_status_sensor
  controller_id: philip
  name: Status
switch.philips_power_switch: [source example.yaml:44]

  Platform not found: 'switch.philips_power_switch'.
  platform: philips_power_switch
  controller_id: philip
  name: Power
  icon: mdi:coffee-maker
```

TillFleisch, 2023-03-29:
I was able to build your yaml snippet on my machine and on a clean Debian container.

I have experienced similar problems with external components on a machine that had Python 3.10 installed.
This may be the case if you're running a native Home Assistant installation since HA has recently updated to 3.10.

The clean Debian container is using Python 3.9.

schlunsen, 2023-03-29:
I'm on a mac m1 with python 3.10 but will try with python 3.9. Thanks for the quick reply! 

I've actually only got introduced to esphome through your project, but I really like the idea of it. Been playing with the esp32 some years ago and trying to get back into it. I'm

Btw. when debugging, where you able to connect via usb to the wemos and power it via that while also connecting to the coffee machine? I hope I can contribute in the coming months with updated info for newcomers.

TillFleisch, 2023-03-29:
Nice! Debugging via USB worked, even when connected to the coffee machine. I recommend using OTA updates/logs.

closed 2023-03-30 by TillFleisch

bkbartk, 2023-11-16:
sorry to bumb this question,
I have to same issue on esphome: 2023.11.1
on Home Assistant OS (I think python 3.11)
is setting up a vm using python 3.9 still a valid work around?

I'm not that familiar with this process.

TillFleisch, 2023-11-16:
@bkbartk , the issue was related to the latest ESPHome addon update. I pushed a fix, which should resolve the compilation issue.

bkbartk, 2023-11-17:
thnx for the quick fix, it works, now I have to wait for the hardware components to arrive.