TillFleisch/ESPHome-Philips-Smart-Coffee #3: Migrate to ESP32 with Ethernet

nighteagle1974, 2023-04-06:
Hello, is there a way to migrate to ESP32 with Ethernet like the WT32-ETH01 module?
I will also use it on a philips ep3246 - so the command may differ?

TillFleisch, 2023-04-06:
Yes, it looks like this is possible. It has little to do with this custom component.
Check out the [ESPHome ](http://esphome.io/) webpage and [GitHub ](https://github.com/esphome/esphome) for more information on supported devices.
[Here ](https://community.home-assistant.io/t/how-i-installed-esphome-on-the-wt32-eth01/359027) someone has managed to get ESPHome running on a device like yours.

The commands on the ep3246 may be different. You can use the [UART Debug](https://esphome.io/components/uart.html#debugging) functionality to listen to messages.

closed 2023-05-02 by TillFleisch

nighteagle1974, 2023-05-02:
Hello again,

so have bought a Philips 5000 Series - i will check if i can debug this traffic from the LCD - it loosk similar the other models. Should i post my experience hier or another place?

TillFleisch, 2023-05-02:
I would recommend opening a new issue with a matching title. Also check out #6  as the mentioned machine has a display as well.

DivanX10, 2023-05-02:
How did I find out that the motherboards of the 2000/3000/5000 series coffee machines are identical, only different control panels, so this connection scheme is the same for everyone, only the transistor is questionable. My screen turns on even without a transistor. The main problem is to catch the data from the management and I can't do it.

[image]

<details>
  <summary><b>Motherboards of coffee machines of different series 1000/2000/3000/5000</b></summary>
  

### Motherboard of the 1000/2000/3000 series coffee machine
[image]
[image]

### The motherboard of the 5000 series coffee machine
[image]
</details>

<details>
  <summary><b>Below I lay out the scheme for the board 1.9.30.330.00._v10</b></summary>
  

### Circuit for the board 1.9.30.330.00._v10
[image]

***

### Communication with the front panel
[image]

***

### Sensors and valves
[image]

***

### Brewing device
[image]

***

### High voltage outputs
[image]

***

### Power supply system
[image]

</details>

I'm trying to figure out from the logs what the command is for turning on coffee brewing and I still can't figure it out. I record by time and see the same records, but which one of them is responsible for the team, I do not see this

[image]

We put ESP Home on Windows, then I created a shortcut, specified the command cmd.exe and the path to esphome
[image]

Copy the project from ESPHome to the ESPHome folder on Windows and run the project
```
esphome logs smart-coffee-philips.yaml > philips5400_protocol.txt
```

Logs are recorded on the computer in a text file
[image]