TillFleisch/ESPHome-Philips-Smart-Coffee #4: Power toggle keeps flickering between on/off

bartbh, 2023-04-13:
To start with: very nice work!

But I'm running into a problem with the 'Power toggle'. With the machine 'OFF', the power toggle in home assistant keeps flickering between 'turned on' and 'turned off'.

I've used your example .yaml with almost no changes, except for a couple of extra buttons for 'SELECT_COFFEE' and 'SELECT_ESPRESSO'.

Part of the logbook for the esp_coffee device:
`April 13, 2023`
`esp_coffee Power turned on  `
`19:58:31 - Now`
`esp_coffee Power turned off`
`19:58:31 - Now`
`esp_coffee Power turned on`
`19:58:29 - 3 seconds ago`
`esp_coffee Power turned off`
`19:58:29 - 3 seconds ago`
`esp_coffee Power turned on`
`19:58:27 - 5 seconds ago`
`esp_coffee Power turned off`
`19:58:27 - 5 seconds ago`

Is this just with my device? I'm using a nodemcu v2 instead of a wemos d1 mini. But the controls in home assistant are working fine, so that doesn't seem to be the problem.

TillFleisch, 2023-04-13:
The state of the power switch is currently determined based on the presences of message going from the display to the mainboard.
The communication between display and Mainboard stops once the machine turns off (on my model).
To me it sounds like on your machine the display is occasionally sending messages even when it's off. This causes the power switch state to turn on.

Can you verify this message behaviour (while the machine is turned off) by observing the communication using the ESPHome UART debug functionality?

Does the communication happen in both directions or is only the display unit sending messages?

What is the model NR. of your machine?

bartbh, 2023-04-14:
I've been using the esp8266 with my Philips 2200 (and 1200) coffee machine before (based on [chris7topher]) and I don't recall seeing status request from the machine when it was off.

ESPHome is new for me, so I'm not a 100% sure if what I did was correct. But I added the following to the configuration

`    
id: uart_display  

    debug:  
      direction: RX  
      dummy_receiver: false  
      sequence:  
        - lambda: UARTDebug::log_hex(direction, bytes, ':');`

With the following result
`[21:18:36][V][component:204]: Component esphome.coroutine took a long time for an operation (0.07 s).
[21:18:36][V][component:205]: Components should block for at most 20-30ms.
[21:18:36][D][uart_debug:114]: <<< 36:46:3A:36:45:1B:5B:30:6D:0D:0A:3B:33:37:6D:5B:56:5D:5B:63:6F:6D:70:6F:6E:65:6E:74:65:6E:74:73:20:73:68:6F:75:6C:64:20:62:6C:6F:63:6B:20:66:6F:72:20:61:74:20:6D:6F:73:74:20:32:30:4B:26:6A:CD:B9:1B:5B:30:6D:0D:0A:65:6E:74:73:20:73:68:6F:75:6C:64:20:62:6C:6F:63:6B:20:66:6F:72:20:61:74:20:6D:6F:73:74:20:32:30:4B:26:6A:CD:B9:1B:5B:30:6D:0D:0A:65:6E:74:73:20:73:68:6F:75:6C:64:20:62:6C:6F:63:6B:20:66:6F:72:20:61:74:20:6D:6F:73:74:20:32:30:4B:26:6A:CD
[21:18:36][V][component:204]: Component esphome.coroutine took a long time for an operation (0.07 s).
[21:18:36][V][component:205]: Components should block for at most 20-30ms.
[21:18:36][D][uart_debug:114]: <<< B9:1B:5B:30:6D:0D:73:74:20:32:30:4B:26:6A:CD:B9:1B:5B:30:6D:0D:1B:5B:30:3B:33:37:6D:5B:56:5D:5B:63:6F:6D:70:6F:6E:65:6E:74:3A:32:30:34:5D:3A:20:43:6F:6D:70:6F:6E:65:6E:74:20:65:73:70:68:6F:6D:65:2E:63:6F:72:6F:75:74:69:6E:65:20:74:6F:6F:6B:20:61:20:6C:6F:6E:67:20:74:69:6D:65:20:66:6F:B9:0A:B9:81:6F:70:65:72:61:74:69:6F:6E:20:28:30:2E:30:37:20:73:29:2E:1B:5B:30:6D:0D:0A:1B:5B:30:3B:33:37:6D:5B:56:5D:5B:63:6F:6D:70:6F:6E:65:6E:74:3A:32:30:35:5D:3A`

So this doesn't look like a D5:55...... code as when the unit is powered on.

TillFleisch, 2023-04-14:
Can you check if the other uart port (mainboard - rx) is busy as well, while the machine is turned off?
Also make sure the ESP and the coffee machine have a common ground.

One possible fix for this issue would be to only check for messages which start with `D5:55`.
If the mainboard uart is silent we can use that one as an indicator.

bartbh, 2023-04-14:
This is the log of the mainboard rx after switching the machine off at 22:38:39

[22:38:39][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:06:26:D5:55:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:06:26
[22:38:39][D][uart_debug:114]: <<< 00
[22:38:40][D][uart_debug:114]: <<< 00
[22:38:41][D][uart_debug:114]: <<< 00
[22:38:42][D][uart_debug:114]: <<< 00
[22:38:43][D][uart_debug:114]: <<< 00
[22:38:45][D][uart_debug:114]: <<< 00
[22:38:46][D][uart_debug:114]: <<< 00
[22:38:47][D][uart_debug:114]: <<< 00

TillFleisch, 2023-04-14:
@bartbh can you check if the changes in the branch `constrain_power_switch_messages` solve the issue?

You can select the branch in your .yaml file like this:
```
external_components:
  - source: github://TillFleisch/ESPHome-Philips-Smart-Coffee@constrain_power_switch_messages
```

bartbh, 2023-04-14:
Yes, that seems to do the trick.  The software switch turns on/off as expected. So turning on the machine by hand turns on the software switch, turning off the machine by sofware switch keeps the switch off.

closed 2023-04-14 by bartbh