TillFleisch/ESPHome-Philips-Smart-Coffee #7: Philips EP3200 Series

sendorm, 2023-04-30:
Hello,

Is there a way to add support for EP3200 series coffee machines. The codes can be found here: 

https://github.com/micki88/Philips-ep3200-testing

Thanks

TillFleisch, 2023-05-02:
Yes, it is possible to add support for the ep3200, but the codes provides in the linked repository only include actions.
The state sensor part of this component may not work, as the ep3200 could use a different messages for status indications.

I would recommended forking this repository and playing around with the codes provided in the linked repository. (Replace the once included in this repo with the new onces)

Adding support for different machines (and subsequent messages/protocols) would require restructuring this repository.
Maybe I will get around to doing that some day, but I cannot promise anything.

sendorm, 2023-05-02:
I am not confident %100 but the linked repo was also a fork of ep2200 and the author stated that everything was working, so statws should be same. I have forked your repo and will give it a go. Thank you.

sendorm, 2023-06-01:
Tried to have a go but it was harder than I anticipated :)

poldim, 2023-06-24:
@TillFleisch I'm working on a local copy of the custom component based on your repo.  As you suggested in your comment above, the status sensor is not working.  It never changes state from off state which means I can't turn off the machine through the ESP as the switch moves back to off state as soon as you turn it on

I'd like to try to see if I can see the messages on the bus to try to solve this.  In your [protocol.md](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/blob/main/protocol.md) document you said "The messages were obtained by listening to the bus." - Can you share how you did this? Was this using the ESP or some external debugger?

It might be nice to have a text sensor that publishes the protocol messages to debug similar to the [MQTT impletementation](https://github.com/micki88/Philips-ep3200-testing/blob/main/withWebserver_EP3200_Philips.ino#L271).  Unfortunately, my C++ skills aren't capable of making it.

TillFleisch, 2023-06-25:
@poldim I would recomment using the built in ESPHome [uart debug functionality](https://esphome.io/components/uart.html#debugging).
Just add the following lines to one of the uart ports.

```
    debug:
      direction: BOTH
      sequence:
        - lambda: UARTDebug::log_hex(direction, bytes, ' ');
```

Publishing all messages via MQTT or a text sensor is not a feasible option, since the bus is always busy.

poldim, 2023-07-04:
> 

> @poldim I would recomment using the built in ESPHome [uart debug functionality](https://esphome.io/components/uart.html#debugging). Just add the following lines to one of the uart ports.
>
> ``` [...]

Awesome, that worked. The semi-processed data is here that just regexed them into individual rows: https://pastebin.com/iuDnQb3S

It appears when I sent the power on command it went through this sequence:
`D5 55 0A 01 02 00 02 00 00 00 0E 12 ` 6x times
`D5 55 02 01 02 00 02 00 00 00 38 15 ` 6x times
^^^ these follow your protocol power on logic

`D5 55 00 01 03 00 12 00 00 00 10 21  ` 5800x times
^^^ these are unexplained as they have `03` and your protocol doesn't address that

`D5 55 00 01 03 00 12 00 00 00 10 C8 `
`D5 55 00 01 03 00 12 00 00 00 10 21`
`D5 55 00 01 03 00 12 00 00 00 88 C8 `
^^^ variants with different checksums on the end but ultimately 
`D5 55 00 01 03 00 12 xx xx xx xx xx` was seen another 252x times 

this is the rest of the data in a pivot table:
[image]

I'll have to run it for longer and see if I can find any pattern, but let me know if you have any other suggestions.

> Publishing all messages via MQTT or a text sensor is not a feasible option, since the bus is always busy.
it appears they solved it by only publishing the value if it was [different than the last](https://github.com/micki88/Philips-ep3200-testing/blob/main/withWebserver_EP3200_Philips.ino#L271), but I'm not sure if that can be implemented in ESPHome
```
  if (strcmp(newStatus, oldStatus) != 0) {
    mqttClient.publish("coffee/status", newStatus);
    strncpy(oldStatus, newStatus, 19);
```

RichyT, 2023-08-27:
That's interesting. Checksums don't change. That's why they're checksums. Also notable is that the 10 repeats once then the C8 is repeated from earlier. Seems like something odd is going on with those two bytes. Could they just be some kind of serialization?

TillFleisch, 2023-08-27:
That 'checksum' problem is interesting.

>Is it confirmed that the 'checksum' doesn't work if it's not correct for the data?

I quickly checked the following on my EP2200:
I've incrased the last byte of the turn on/off and make coffee button commands by 1. The coffee machine did not respond as expected(nothing happend). Logs showed a `00 .. 00` empty display response. In my case the machine is reading the messages and responding but not interpreting, since the checksum is wrong.

The messages seem to be very consistent. The checksums remains unchanged.
In between lots of these (`D5 55 00 01 02 00 02 00 00 00 11 36`) messages I found:
`D5 55 00 01 20 02 00 00 00 11 36` which seems to be currupted. I presume the mainboard just ignored it.

I guess @poldim could be experiencing noise on the bus. It is wierd though, that only the last bytes are affected. This may be a phenomenon caused by filtering the messages starting with `D5 55`. I would suggest re-check wiring and proper ground connections.

RichyT, 2023-08-29:
Noise would also help explain the apparent high number of what I would assume are retries.

sepodele, 2023-08-29:
Hm. On my 3200 (EP3243) series I did get logs saying On/Off and Cleaning when it started to clean. Without customizing any code I think (I've now tried replacing the codes for the 2200 with 3200 from https://github.com/mkorenko/esp-phillips-3200/ ). But my problem is that my ESP32-S2 Mini won't boot if power pin is plugged in. I can plug it in after boot and turning on the machine works from HA and from display button and I can use the action to make a coffee from HA. But the display doesn't work even though there are lights on it. Does this look correctly wired to you? 
[image]

sepodele, 2023-12-06:
I replaced the S2 Mini with a Wemos D1 Mini and I'm now using a MOSFET instead. I've used debug logging to find pre-power, power with cleaning and power without cleaning. But when I turn it on with cleaning (either from HA or from the display button) the lights on the display go on normally in a circle but right when the water is supposed to come out there's a beep and it stops.

What could cause that? It won't finish the cleaning cycle.

sepodele, 2023-12-07:
And when I hit 'Select Coffee' it selects coffee but starts by itself after 2 seconds but then stops immediately with a beep. I don't understand why it starts by itself and I don't understand why it stops.

TillFleisch, 2023-12-08:
> there's a beep and it stops

That sounds like the power off button is pressed (either hardware/stimulated).
I would recommend logging the messages and looking for power off messages. Maybe the ESPHome button component is turned off (possibly by accident, or timeout (since the machine status is not detected)), which would result in this behaviour.

> when I hit 'Select Coffee' it selects coffee but starts by itself after 2 seconds but then stops immediately with a beep

The starting by it self could be a configuration problem. Based on the selected action, the play button is pressed automatically (actions stating with `make_`). As for the stopping, I have no clue but would recommend logging. Maybe this can give more insight. The beep sound indicates that a button was pressed (possibly play/pause which would abort the process).

sepodele, 2023-12-09:
Maybe the Wemos D1 Mini I have is just weird. I have the USB-C variant. I went back to trying my ESP32 S2 Mini again and there it seems like everything works except for a consistent power on (which was reliable on the D1 Mini)... I think I need to use inverted = true but can't find a GPIO PIN working consistently. After the machine has been off for a while and I turn it on in HA the display lights up and works like it should. But if I turn it off and on after 10 seconds the display remains off. I've tried changing the default 1s to 0.3s, 0.5s and 1.5s and nothing seems to work reliably. Ideas?

TillFleisch, 2023-12-09:
Sometimes, on my machine the display does not turn on (but the machine does), when turned on through HA.
Every once in a while, I wake up to the machine appearing to be turned off. If I press the power button, the machine is available immediately.
I have not figured out what the cause for this is. From a different project I know that the UART timing (after using flush) is weird ever since the ESPHome update that updated the Arduino core version to 2.0+. (The time between calling flush and the bus being written to is longer and inconsistent). I'm trying various different things, but due to the randomness of this problem, troubleshooting this issue takes a bit.

sepodele, 2023-12-10:
Current state: When power to the machine has been off I can't turn it on with the power button on the display. I have to use HA to turn it on. Then the display comes on normally. If I then turn off the machine (either from display or from HA), I can only turn on the machine (and display) from the display power button. Turning it on from HA makes the machine turn on but not the display. It doesn't matter if I wait 1 second or 10. But like you said, touching the power button makes the display come on immediately.

Current settings on my ESP32 S2 Mini:
  power_pin: GPIO34
  invert_power_pin: true
  power_trip_delay: 1s

sepodele, 2023-12-11:
This is my wiring atm:

[image]

TillFleisch, 2024-01-07:
> Sometimes, on my machine the display does not turn on (but the machine does), when turned on through HA.

The latest changes seem to have resolved this issue for me.  The ESP will now perfrom multiple power trips, if the display does not turn on.
Maybe this helps with your problem as well.

sepodele, 2024-01-11:
> [19:23:00][W][component:214]: Component esphome.coroutine took a long time for an operation (1.00 s).
[19:23:00][W][component:215]: Components should block for at most 20-30ms.
[19:23:02][W][component:214]: Component esphome.coroutine took a long time for an operation (1.00 s).
[19:23:02][W][component:215]: Components should block for at most 20-30ms.
[19:23:04][W][component:214]: Component esphome.coroutine took a long time for an operation (1.00 s).
[19:23:04][W][component:215]: Components should block for at most 20-30ms.
[19:23:06][W][component:214]: Component esphome.coroutine took a long time for an operation (1.00 s).
[19:23:06][W][component:215]: Components should block for at most 20-30ms.
[19:23:08][W][component:214]: Component esphome.coroutine took a long time for an operation (1.00 s).
[19:23:08][W][component:215]: Components should block for at most 20-30ms.
[19:23:10][W][component:214]: Component esphome.coroutine took a long time for an operation (1.00 s).
[19:23:10][W][component:215]: Components should block for at most 20-30ms.
[19:23:11][E][philips_power_switch:024]: Power tripping display failed!

Same behaviour unfortunately. Display will only turn on from HA if power to the machine and ESP32 has been removed completely.

sepodele, 2024-01-11:
Actually, if I just remove the power from VBUS (red cable) or push the RST button so that the ESP32 reboots it IS able to turn on the display each time. Any ideas why?

TillFleisch, 2024-01-11:
I am pretty clueless. Maybe there is some way for current to flow and the mosfet 'removing' the ground pins isn't enough for the display to restart.

sepodele, 2024-01-11:
I got a good suggestion at ESPHome on Discord. I replaced:

> power_pin_->digital_write(!power_pin_->digital_read());
> delay(power_trip_delay_);
> power_pin_->digital_write(!power_pin_->digital_read());

With:

> power_pin_->digital_write(false); // Turn off
> delay(power_trip_delay_);
> power_pin_->digital_write(true); // Turn on

And removed 'invert_power_pin: true'.

Then it seems to work every time. There's one weird issue though. If I cut the power to the machine and the ESP and turn it back on, it beeps after 11 seconds, then after 8 seconds and display comes on with lights on every drink and a blinking Aqua Clean. The display is not responding to touch and status in Home Assistant is that it's turned on and 'Cleaning'. Workaround seems to be to turn the machine on immediately after power plug is connected.

TillFleisch, 2024-01-12:
> power_pin_->digital_write(false); // Turn off
delay(power_trip_delay_);
power_pin_->digital_write(true); // Turn on

This should also happen with the pin inversion method if you use `invert_power_pin: false` (which is the default configuration).

The turning on after power-up is weird and may be a side effect from using a strapping pin. See [this documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html) for more information regarding strapping pins on the ESP32. Same goes for the 8266.

sepodele, 2024-01-12:
> This should also happen with the pin inversion method if you use `invert_power_pin: false` (which is the default configuration). Can you verify that this is working as well?

No. Just tested again. When I use 

> power_pin_->digital_write(!power_pin_->digital_read());
delay(power_trip_delay_);
power_pin_->digital_write(!power_pin_->digital_read());

And 

>   invert_power_pin: false 

It's not turning on the display.

>

TillFleisch, 2024-01-12:
Hm.. it seems inverting the power pin isn't a valid move after all. I'll open up a pr to close this issue.

Are you using the Arduino or IDF Framework?

sepodele, 2024-01-12:
Arduino?

`esp32:
  board: lolin_s2_mini
  framework:
    type: arduino`

sepodele, 2024-01-13:
Btw, what will be the best way to update the status sensor with "Brewing" for instance after start button is pressed? I'd like to create some kinda logging of each coffee typed brewed. The 5400 series has that in the display but we could do it in HA. :) If we add 'Brewing' to status sensor I guess an automation in HA triggering on "to state 'Brewing'" and logging the from state could work. Maybe you have a better suggestion?

And what's your reasoning behind having one size adjustment and one beans adjustment for each coffee type? Simplicity? Wouldn't it make more sense to have just one size adjustment and one beans adjustment and it will work for every coffee type? My 3200 series has Coffe, Espresso, Americano, Cappuccino and Latte Macchiato.

I'm gonna fork this repo soon (when I can find time) for my EP3200 version. I added milk level settings and it seems like everything is working. I haven't figured out a good way of selecting ground coffee though. Not sure if a 4th level to size is the way too go.

TillFleisch, 2024-01-14:
>Btw, what will be the best way to update the status sensor with "Brewing" for instance after start button is pressed? I'd like to create some kinda logging of each coffee typed brewed.

For a quick solution I would recommend using template sensors and global variables in ESPHome directly. This way you could remember the last selected coffee type and then update your template sensor accordinly, once it goes to brewing. Keep in mind that the state change to `[x] Selected` must be awaited, since it is not always visible when using the `make_coffee` option on the button component.

>And what's your reasoning behind having one size adjustment and one beans adjustment for each coffee type? 

Your suggestion would work just as well. I added them per-type, such that configuration is easier. I **think** building automations with these components is easier as the 'available/unavailable' status can be used as a trigger, which results in less nested/complex yaml configurations.
If you want to build a single input for these, I believe you can set the individual number component to `internal: true` and merge them together using template components.

> I haven't figured out a good way of selecting ground coffee though.

I would say a button component would be best suited for this application. I decided not to add this as a feature, as there is little/no point in automating this functionality. The assumption being: If someone needs to use the group coffee function, they have to interact with the machine anyways.

> I'm gonna fork this repo soon (when I can find time) for my EP3200 version.

Happy coding!

mikulik86, 2024-01-19:
> I would say a button component would be best suited for this application. I decided not to add this as a feature, as there is little/no point in automating this functionality. The assumption being: [...]

I'm thinking about using ground coffee function as an "advanced cleaning". I plan to build in an automatic water filling and a drip tray drainage. I always thought that the cleaning cycle is not sufficient, but with limited amount of water it have to do. Although when water supply is not an issue, I'd like to run the ground coffee sequence without the actual ground coffee to clean the machine before power off.

quack3d, 2024-01-22:
> For a quick solution I would recommend using template sensors and global variables in ESPHome directly. This way you could remember the last selected coffee type and then update your template sensor [...]

I ended up just renaming all the "Busy" states to "Brewing Coffee", "Brewing Americano" etc. Then some kinda template sensor in HA could sum state changes to log it. However, a more proper way would be like you suggested with template sensors directly in ESPHome. I don't want to create one component for each brew though. Maybe you'd like to add it if you find time to refactor the file structure? :) https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/issues/21 

Here is my 3200 Series fork btw: https://github.com/quack3d/ESPHome-Philips-Smart-Coffee-3200-Series

quack3d, 2024-02-04:
Default any source is neat. Thanks for that. Btw, is 'model(Optional: int)' an indication that you plan to add support for 3200 series as well? I merged your refactoring to my fork but now I can't figure out how to implement the milk button. I'm struggling with philips_beverage_setting and the if(is_bean) - press_bean else press_size. How do I add milk level too? Milk is data[13]. And selecting it is ({0xD5, 0x55, 0x00, 0x01, 0x03, 0x00, 0x0E, 0x00, 0x08, 0x00, 0x1F, 0x16});

TillFleisch, 2024-02-04:
> Btw, is 'model(Optional: int)' an indication that you plan to add support for 3200 series as well?

No. I don't own a 3200, nor do I have access to one. :/ This change is supposed to make supporting it easier. We'd also need to change the base component name to make things less confusing, which is a breaking change.

> if(is_bean)

I'm assuming that the 3200 series has 3 size selections things? In that case the beverage settings component would need to be refactored to support more than 2 types. What other differences are there between the EP2220 and the 3200?

quack3d, 2024-02-04:
Yeah, 3 differences for milk, same as beans and size. It has Americano and Latte Macchiato in addition to the ones on 2200 series. Use_cappuccino makes it select Latte Macchiato so in my fork I just changed it to use_latte.

quack3d, 2024-02-05:
All of these in my fork work: [commands.h](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/blob/main/components/philips_series_2200/commands.h) I think I also tried the coffeePulver from your link but forgot to add it. 

Your status sensor works quite well. I just had to modify the 'idle' status (got the info from [here](https://github.com/flashmandv/phillips-3246-smart/blob/main/coffee_machine/src/esp-phillips-3200/esp-phillips-3200.cpp#L333-L344)):

`if (data[3] == led_on && data[4] == led_on && data[5] == led_on && data[6] == led_on)`
to
`if (data[3] == led_on && data[4] == led_on && data[5] == led_on && data[13] == led_off && data[14] == led_off && data[15] == led_off)`

And I added Americano and Cappuccino since your steam alternate makes Latte Macchiato on my 3200 series. :)

TillFleisch, 2024-02-05:
|  | Byte | Purpose           | Detail                                                |
| - | ---- | ----------------- | ----------------------------------------------------- |
|| 0    | START             |
|| 1    | START             |
|| 2    | INSTRUCTION       |
|| 3    | Espresso-LED      | `03`/`07` - half/full brightness ; `38` - 2x espresso          |
|x| 4    | ~~Hot Water-LED~~ *Cappuccino-LED*     | `03`/`07` - half/full brightness                               |
|| 5    | Coffee-LED       | `03`/`07` - half/full brightness; `38` - 2x coffee             |
|x| 6    | ~~Steam-LED~~  *Latte/Americano-LED*        | `03`/`07` (half/full) - latte selected; `38` - americano slected|
|x| 7    |   *Hot Water/ 2x Americano (Series 3200 only)*                |  `07` - 2x Americano; `38` - Hot Water selected |
|| 8    | Bean-LED          | `00` - 1 LED; `38` - 2 LEDs; `3F` - 3 LEDs                     |
|| 9    | Bean-LED          | `07` - show led group; `38` - powder selected                  |
|| 10   | Size-LED          | `00` - 1 LED; `38` - 2 LEDs; `3F` - 3 LEDs; `07` -  Top LED    |
|| 11   | Size-LED          | `07` - show led group                                          |
|x| 12   | *Milk-LED*     | `00` - 1 LED; `38` - 2 LEDs; `3F` - 3 LEDs                                 |
|x| 13   |*Milk-LED*      | `07` - show led group                                 |
|| 14   | Waste&Warning-LED | `07` - waste; `38` - warning sign                              |
|| 15   | Play/Pause-LED    | `07` - on                                                      |
||16   | checksum          |
|| 17   | checksum          |

This is my first guess at the communication schema based on the provided information. (`x` denotes rows with differencese from the EP2220 schema) This information is not verified and I cannot verify it.

TillFleisch, 2024-02-05:
I was goofing around a little bit just for fun. This is entirely untested, missing documentation and probably contains bugs.

quack3d, 2024-02-07:
Awesome! Everything except for milk seems to work. This in beverage_setting.cpp seems to fix milk also:
```
                    //uint8_t enable_byte = type_ == BEAN ? 9 : (type_ == SIZE ? 11 : 13);
                    //uint8_t amount_byte = type_ == BEAN ? 8 : (type_ == SIZE ? 10 : 12);
		    uint8_t enable_byte = type_ == BEAN ? 9 : (type_ == SIZE ? 11 : 11);
                    uint8_t amount_byte = type_ == BEAN ? 8 : (type_ == SIZE ? 10 : 13);
``` 
If you fix that the only thing missing from your version (for me) is showing which brew was made so I can log stuff. 😊

TillFleisch, 2024-02-07:
> showing which brew was made so I can log stuff.

This is a story for a different PR and it's also a breaking change (if it's the new default). I could also make it an option (choose between `busy` and `brewing [...]`). If I recall correctly, there were some invalid/incorrect states... This need some double checking.

> ```c++
> uint8_t enable_byte = type_ == BEAN ? 9 : (type_ == SIZE ? 11 : 11);
> uint8_t amount_byte = type_ == BEAN ? 8 : (type_ == SIZE ? 10 : 13);
```

This seems weird. Is the milk selection Option always available when size can be selecte. (In other words: is there a configuration in which only milk can be adjusted and not size).
Would you mind generating a log (motherboard uart both directions) for 

|  | Byte | Purpose           | Detail                                                |
| - | ---- | ----------------- | ----------------------------------------------------- |
|| 0    | START             |
| Byte | Purpose           | Detail                                                |
| ---- | ----------------- | ----------------------------------------------------- |
| 13   |*Milk-LED*         | `07` - show led group                                 |

quack3d, 2024-02-07:
> I could also make it an option (choose between busy and brewing [...]). If I recall correctly, there were some invalid/incorrect states... This need some double checking.

That would be neat. I haven't experience any invalid/incorrect states but I never used the MAKE actions. I only use the SELECT ones. I did however notice now that when I long pressed the bean button the status went to 'unknown' briefly before it correctly showed Ground Cofffee selected.

>is there a configuration in which only milk can be adjusted and not size

No. Milk lights only show when Cappuccino or Latte Macchiato is selected. And both can be adjusted in size, bean and milk.

I'll try creating a log like you suggested.

quack3d, 2024-02-07:
Does it matter if I did it from HA instead of manually pressing buttons? Here is the log when using HA buttons: https://pastebin.com/GSJiAKL6

I guess you can see it in the logs but this is what I did: Turned on without clean, waited until Idle, chose Coffee, chose Cappuccino and then milk level 1.

quack3d, 2024-02-07:
LATTE_MACCHIATO is also missing from beverage_setting.cpp btw. Works when I add :
```
 ((source_ == LATTE_MACCHIATO || source_ == ANY) &&
 status_sensor_->get_raw_state().compare("Latte Macchiato selected") == 0))
```

TillFleisch, 2024-02-07:
Looks like byte `12` is always `00`. Here is the updated command overview:

### EP3234

| Byte | Purpose                    | Detail                                                            |
| ---- | -------------------------- | ----------------------------------------------------------------- |
| 0    | START                      |                                                                   |
| 1    | START                      |                                                                   |
| 2    | INSTRUCTION                |                                                                   |
| 3    | Espresso-LED               | `03`/`07` - half/full brightness ; `38` - 2x espresso             |
| 4    | Cappuccino-LED             | `03`/`07` - half/full brightness                                  |
| 5    | Coffee-LED                 | `03`/`07` - half/full brightness; `38` - 2x coffee                |
| 6    | Latte/Americano-LED        | `03`/`07` (half/full) - latte selected; `38` - Americano selected |
| 7    | Hot Water/ 2x Americano    | `07` - 2x Americano; `38` - Hot Water selected                    |
| 8    | Bean-LED                   | `00` - 1 LED; `38` - 2 LEDs; `3F` - 3 LEDs                        |
| 9    | Bean-LED                   | `07` - show led group; `38` - powder selected                     |
| 10   | Size-LED                   | `00` - 1 LED; `38` - 2 LEDs; `3F` - 3 LEDs; `07` -  Top LED       |
| 11   | Size-LED                   | `07` - show led group (size and milk)                             |
| 12   | Unknown                    |  probably aqua_clean/calc_clean                                   |
| 13   | Milk-LED                   | `00` - 1 LED; `38` - 2 LEDs; `3F` - 3 LEDs                        |
| 14   | Waste&Warning-LED          | `07` - waste; `38` - warning sign                                 |
| 15   | Play/Pause-LED             | `07` - on                                                         |
| 16   | checksum                   |                                                                   |
| 17   | checksum                   |                                                                   |

TillFleisch, 2024-02-07:
Do cappuccino and latte have a `2x` option?

quack3d, 2024-02-07:
No 2x option for Cappuccino or Latte. It's because the milk can only hit one cup.
Ground coffee is an option for Americano, yes.
Btw, the log above should have a blinking aqua clean somewhere in there. The light started blinking yesterday and I haven't changed the filter yet. Not that it's important to monitor for that light but in case you were curious. :)

machintrucbidule, 2024-02-07:
Hi,

I own a Philips series 3200, but a 3221 rather than a 3243. Meaning that options are close to the 3243, but with some differences as I don't have the milk/latte options. So current code for 3200 won't work with mine I gues.
Dashboard looks like this: 
[image]

On left coffee part I have: Expresso, Expresso lungo, Coffee, Americano, Hot Water, Steam.
And on the right part: Bean, Size, Temperature, and then Aqua Clean and Clean.

I ordered hardware to build this (still waiting for it), but would you be able to tell me how to adapt the code for my machine? How identify which command is sent when pressing a button, and what should I update in the code in order to add compatibility with my 3221 while not breaking 3243 one?

Thanks!

TillFleisch, 2024-02-07:
@machintrucbidule: Please open a separate issue for the EP3221.

TillFleisch, 2024-02-07:
> Btw, the log above should have a blinking aqua clean somewhere in there. The light started blinking yesterday and I haven't changed the filter yet. Not that it's important to monitor for that light but in case you were curious. :)

I just bought stuff for calc-cleaning my machine. Maybe, I will log some Messages to figure out where the calc/clean led is encoded :)

I have added your suggestions to the PR draft.

quack3d, 2024-02-10:
Thanks. It works well. The only "logic issue" now is that Bean becomes unavailable when selecting Hot water but Milk does not. Milk is also available when selecting Coffee or Americano. Doesn't really matter though. 😋

TillFleisch, 2024-02-10:
> Milk is also available when selecting Coffee or Americano

It's also available when selecting espresso, which it shouldn't? Thank you for pointing this out.
It matters to me and it's a quick and simple fix.

quack3d, 2024-02-10:
Yeah. Shouldn't be available when selecting Espresso.

sendorm, 2024-02-10:
> I was goofing around a little bit just for fun. This is entirely untested, missing documentation and probably contains bugs. You can play around with this using (yaml config change required):
> 
> ```yaml
> external_components:
>   - source: github://TillFleisch/ESPHome-Philips-Smart-Coffee@series-3200
> ```
> 
> I should stop procrastinating and should continue learning for my exams

I’ve started using this with my ep3246/50 and everything is working fine. Thank you everyone dor their hard work. 
[image]
[image]
[image]

TillFleisch, 2024-02-10:
Is there any difference between the two? If not I would add the `EP3246` as an alias for the EP3243.

sendorm, 2024-02-10:
We have reached a conclusion with quack3d that the differences are only cosmetic. 3243 is white while 3246 is black.

sendorm, 2024-02-10:
One thing to add though latte machiato button does not work on mine. Also the clean with and without water work at the same time and it always dispenses water at first.

TillFleisch, 2024-02-10:
@quack3d can you replicate this behaviour?

@sendorm can you generate a log containing the button presses (when using the buttons on the display)? The power/clean-power commands can also be retreived this way.  My EP2220 always performs a cleaning cycle after being unplugged. This way I retrieved the power on command with cleaning. The without cleaning command can be retrieved by powering on the machine, powering it off and then powering it on again (always using the display button). The machine should skip the cleaning cycle since it has already cleaned and it's sill pre-heated.

When using both power switches, both are supposed to reflect the state of the machine (turn on). The Action they perform is different.

quack3d, 2024-02-10:
I had a dialog with sendorm and it seems he's experiencing what you wrote about the machine always cleaning if it has been unplugged. It's the same with my machine.

But the command we've used for Latte ({0xD5, 0x55, 0x00, 0x01, 0x02, 0x00, 0x02, 0x10, 0x00, 0x00, 0x09, 0x26}) was more or less accidentally discovered in that it selected Latte when I tried your command for cappuccino. I never pressed the Latte button to figure out the 'real' command. But I now tested the one from other repos (which you also linked to previously): {0xD5, 0x55, 0x00, 0x01, 0x03, 0x00, 0x0E, 0x10, 0x00, 0x00, 0x2D, 0x24}; And that also works on my EP3243. So perhaps that is the command intended for the 3200 series? Perhaps you should change to the one ending with 0x24 and @sendorm can test as well?

sendorm, 2024-02-11:
I've checked the debug logs and after pressing the latte button this code seems to repeat itself:
D5 55 00 00 00 07 00 00 38 07 38 07 00 00 00 00 07 33 15 on 3246.

TillFleisch, 2024-02-11:
> I've checked the debug logs and after pressing the latte button this code seems to repeat itself:
D5 55 00 00 00 07 00 00 38 07 38 07 00 00 00 00 07 33 15 on 3246.

Sorry, I should have clarified. I meant messages from the display to the mainboard (RX on the display uart or TX on the mainboard uart).

> 0xD5, 0x55, 0x00, 0x01, 0x03, 0x00, 0x0E, 0x10, 0x00, 0x00, 0x2D, 0x24

This seems to be in line with [this resource](https://github.com/flashmandv/phillips-3246-smart/blob/b5276750fca483eccd3042b9c13bd59696ed50f7/coffee_machine/src/esp-phillips-3200/esp-phillips-3200.cpp#L5C1-L54C3). The Espresso, aqua clean and calc clean button commands also seem to be wrong. I have replaced the current ones with the ones from the linked source.

sendorm, 2024-02-11:
> Sorry, I should have clarified. I meant messages from the display to the mainboard (RX on the display uart or TX on the mainboard uart).
>
>
> This seems to be in line with [this [...]

Well it still doesn't apply from homeassistant side. But the status changes to "Latte Macchiato selected" when pressed from the machine.

The log from the other side is this (when latte button is pressed from machine side)
[11:47:01][D][text_sensor:064]: 'Status': Sending state 'Latte Macchiato selected'
[11:47:01][D][uart_debug:114]: >>> 38 07 00 00 19 0A D5 55 00 00 00 00 07 00 38 07 38 07 00 38 07 00 00 19 0A
[11:47:01][D][uart_debug:114]: <<< 34 D5 55 00 01 03 00 0E 00 00 00 35 34 D5 55 00 01 03 00 0E 00 00 00 35 34 D5 55 00 01 03 00 0E 00 00 00 35 34 D5 55 00 01 03 00 0E 00 00 00 35 34 D5 55 00 01 03 00 0E 00 00 00 35 34 D5 55 00
[11:47:01][D][uart_debug:114]: >>> D5 55 00 00 00 00 07 00 38 07 38 07 00 38 07 00 07 20 16

TillFleisch, 2024-02-11:
The message ending on `35 34` seems to be the request update/idle messages (when no button is pressed).

You may need to add the `refresh` key when adding the external component (see [documentation](https://esphome.io/components/external_components.html#external-components)) as ESPHome may still be using a older cached version of the repo.

[30 more replies]