TillFleisch/ESPHome-Philips-Smart-Coffee #8: Display stops working when connected mosfet

isocron, 2023-05-16:
Hello.
Hope someone can help me.
I got a Philips Series 2200 and plugged in a Wemos D1 Mini. I didn't have a mosfet at the time, so I connected ESP without it.
Everything works, except turning it on from Homeassistant as expected.
Now I have found a mosfet, IRFZ44N, and plugged it in, but now it doesn't work as expected anymore.

From the display:
I can turn on the power to the coffee maker.
It starts and cycles through the icons. When it's done, only two random icons light up, not all four, and no buttons work.

From Home assistant:
I can turn on the device by double-triple clicking on power.
I can select product/strength/size and start brewing but the display on the coffee maker shows nothing.
Is the IRFZ44N the wrong choice? Anyone know one that works?

DivanX10, 2023-05-16:
If you return to the original position, how will the control panel work? Remove the ESP and connect the wires as it was. This is necessary to understand who is the culprit, is the coffee machine itself broken or is it still because of esp

TillFleisch, 2023-05-16:
I used a IRLZ44N which has a lower gate thershold voltage. The 3.3v provided by the ESP may not be enough to fully open the gate on the IRFZ44N which could cause weird behavior.
Can you confirm the display panel is getting 5V?

DivanX10, 2023-05-16:
Now I have connected a multimeter and am attaching a photo. As you can see, 4,993V, which is equal to 5V

[image]

isocron, 2023-05-16:
Didn't even think about it. I'll check when I get back home.

DivanX10, 2023-05-26:
> Hi again. Sorry for late post.
> I did check with multimeter and the volatage out is 3v to the display when im using the IRFZ44N.
> I have orderd an IRLZ44N but it's delayd in the postsystem. Will give som more info when I tested it.

Measure the voltage without a transistor, it should be +5V. See a transistor that has a threshold gate voltage of +3v, for example a [BS170 transistor](https://pl-1.org/getproductfile.axd?id=9488&filename=BS170.pdf).

[image]

TillFleisch, 2023-07-03:
@isocron did the IRLZ44N solve the problem? Can we close this issue?

isocron, 2023-07-03:
@TillFleisch sorry, yes that problem is solved. Thank you!