TillFleisch/ESPHome-Philips-Smart-Coffee #10: UART prblems

davidepanato, 2023-05-25:
Hello, first of all thank you for your work, it's awesome!

I've tried t setup your project on my EP2200 but I'm facing some problems on UART behavior.
I have checked the HW and looks fine because I can both tx and rx data on the interfaces.

Unfortunately it seems that it is not working because I see a constant flow of data on the UART and web interface is not detecting any state change in the coffe macchine.

In the logs I can see (mainboard logs):

19:51:39 | [D] | [uart_debug:114] | >>> 00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01
-- | -- | -- | --
19:51:40 | [D] | [uart_debug:114] | <<< 00:00:00:00:06:26:D5:55:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:06:26
19:51:40 | [D] | [uart_debug:114] | >>> D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00
19:51:40 | [D] | [uart_debug:114] | <<< D5:55:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:06:26
19:51:40 | [D] | [uart_debug:114] | >>> 00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00
19:51:40 | [D] | [uart_debug:114] | <<< D5:55:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:06:26:D5:55:01:00:00:00:00:00:00:00:00:00:00
19:51:40 | [D] | [uart_debug:114] | >>> **00:03:02:00:00:00:2D:01**:D5:55:00:00:00:03:02:00:00:00:2D:01:**D5:55:00:00:00:03:02:00:00:00:2D:01**
19:51:40 | [D] | [uart_debug:114] | <<< 00:00:00:00:06:26:D5:55:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:06:26
19:51:40 | [D] | [uart_debug:114] | >>> D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00
19:51:40 | [D] | [uart_debug:114] | <<< D5:55:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:06:26
19:51:40 | [D] | [uart_debug:114] | >>> 00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00
19:51:40 | [D] | [uart_debug:114] | <<< D5:55:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:06:26:D5:55:01:00:00:00:00:00:00:00:00:00:00
19:51:40 | [D] | [uart_debug:114]

For some reason it looks like the data that it's receiving are not "aligned".

My system is base on an esp8266 amica nodemcu.
My yaml is exactly same as yours except for few changes:

Added log printing for mainboard UART
uart:
  # UART connected to the mainboard
  - tx_pin: GPIO1
    rx_pin: GPIO3
    baud_rate: 115200
    id: uart_mainboard
    debug:
      direction: BOTH
      dummy_receiver: false
      after:
        delimiter: "\n"
      sequence:
        - lambda: UARTDebug::log_hex(direction, bytes, ':');

added webserver
web_server:
  port: 80

Any idea why I see this behavior?
Thank you!
Dave

TillFleisch, 2023-05-25:
>data that it's receiving are not "aligned"

This is expected behavior. The mainboard is continuously sending updates while the machine and display are turned on. The debug component will be out of sync with the start bytes of the messages since they have different lengths.

It looks like the wiring is correct. The mainboard is sending the following command: `D5:55:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:06:26`
This command is partially responsible for turning on the display unit.
If the display is not turned on properly (no lights) the mainboard will not send further updates and the status sensor will not work.
This may be your problem if you are only seeing the messages posted above.
Instead of all `00` you should see a message similar to this one: `D5:55:00:07:07:07:07:00:00:00:00:00:38:00:00:00:00:37:28`.

I would recommend trying the following things:
1. Check if the display unit is on (LEDs are on)
2. Check if the machine works as usual when using the buttons on the display
3. Try running the esp without UART debug
    - Note that the `delimiter: "\n"`  will not work as `\n` is never communicated on the bus
    - The UART debug component can introduce latency which may have undesired side effects

davidepanato, 2023-05-26:
Hello Tiil,
  I tried your suggestion and removed all debug functionalities and web server to be 100% aligned to your implementation but it still not work.
Without the adapter everything works perfectly in the machine.
What is weird is that when esp device is connected I can power on the coffee machine from external display.
After that things starts to be weird because the led loop during cleanup process is not "smooth" (sometimes one led is not powered on and timing is not correct).
After the cleaning procedure is completed external display is not working anymore, leds are ON but button pressure is not detected.
When I try to send commands from home assistant I can hear bips from the machine but int he end nothing really happens.
Any idea on how to proceed to debug the problem?

Dave

DivanX10, 2023-05-26:
It is best to connect ESP32, since ESP32 has 2 hardware UART, which can better capture data and does not hang like esp8266.

[image]

Bytes will not be aligned during debugging, if you want to align them, then you need to know the size of the bytes and specify them. I counted at `D5:55:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:06:26` 19 bytes. To output even bytes, we specify after: bytes in the config: 19. After that, your bytes will be even, but this may distort the reliability of the data received, since the length of the bytes may be different, so it's better not to specify this parameter, but you can try

```
after:
  bytes: 19
```

The full code looks like this
```
 - id: uart_mainboard
   rx_pin: GPIO3
   tx_pin: GPIO1
   baud_rate: 115200 #The speed is determined by the oscilloscope and analyzer
   stop_bits: 1 #Determined by the analyzer. that there is 1 stop bit
   data_bits: 8 #Determined by the analyzer. that there are 8 bits of data
   parity: NONE #It is determined by the analyzer that there is no parity and odd, therefore strictly NONE
   rx_buffer_size: 256
   debug:
     direction: TX #I view only TX messages from the motherboard to the control panel
     dummy_receiver: false #We turn off emulation, since we control a real device 
     after:
       bytes: 19 #Specify the number of bytes displayed
```

For debugging, I use this option, where I turned off BOTH and RX. Specified only TX
```
uart:
 - id: uart_display
   rx_pin: GPIO16
   tx_pin: GPIO17
   baud_rate: 115200
   stop_bits: 1
   data_bits: 8
   parity: NONE
   rx_buffer_size: 256
   debug:
     direction: TX
     dummy_receiver: false

 - id: uart_mainboard
   rx_pin: GPIO3
   tx_pin: GPIO1
   baud_rate: 115200
   stop_bits: 1
   data_bits: 8
   parity: NONE
   rx_buffer_size: 256
   debug:
     direction: TX
     dummy_receiver: false
```

<details>
  <summary>After that, I see logs in this format. I will note that I have a 5000 series and the logs are different from the 2000 series, but just for understanding I will show them</summary>
  

18:21:49	[D]	[uart_debug:114
00:00:16:8A:15:07:55:AA:AA:AA:B0:CF:07:06:00:00:00:00:00:00:F5:13:17:63:55:AA:AA:AA:B0:D0:07:06:00:00:00:00:00:00:55:9E:6E:10:55:AA:AA:AA:B0:D1:07:06:00:00:00:00:00:00:16:8A:15:07:55:AA:AA:AA:B0:CF:07:06:00:00:00:00:00:00:F5:13:17:63:55:AA:AA:AA:B0:D0:07:06:00:00:00:00:00:00:55:9E:6E:10:55:AA:AA:AA:B0:D1:07:06:00:00:00:00:00:00:16:8A:15:07:55:AA:AA:AA:B0:CF:07:06:00:00:00:00:00:00:F5:13:17:63:55:AA:AA:AA:B0:D2:07:06:00:00:00:00:00:00:D3:B6:98:3E

18:21:50	[D]	[uart_debug:114]	
55:AA:AA:AA:FF:CF:01:CF:F0:E6:72:77:55:AA:AA:AA:FF:CF:01:CF:F0:E6:72:77:55:AA:AA:AA:FF:CF:01:CF:F0:E6:72:77:55:AA:AA:AA:FF:CF:01:CF:F0:E6:72:77:55:AA:AA:AA:FF:D0:01:D0:48:0F:00:ED:55:AA:AA:AA:FF:D0:01:D0:48:0F:00:ED:55:AA:AA:AA:FF:D0:01:D0:48:0F:00:ED:55:AA:AA:AA:FF:D0:01:D0:48:0F:00:ED:55:AA:AA:AA:FF:D0:01:D0:48:0F:00:ED:55:AA:AA:AA:FF:D0:01:D0:48:0F:00:ED:55:AA:AA:AA:FF:D0:01:D0:48:0F:00:ED:55:AA:AA:AA:FF:D1:01:D1:E9:55:C5:9B:55:AA:AA:AA:FF:D1

18:21:50	[D]	[uart_debug:114]	
55:AA:AA:AA:B0:D0:07:06:00:00:00:00:00:00:55:9E:6E:10:55:AA:AA:AA:B0:D1:07:06:00:00:00:00:00:00:16:8A:15:07:55:AA:AA:AA:B0:D0:07:06:00:00:00:00:00:00:55:9E:6E:10:55:AA:AA:AA:B0:D2:07:06:00:00:00:00:00:00:D3:B6:98:3E:55:AA:AA:AA:B0:D1:07:06:00:00:00:00:00:00:16:8A:15:07:55:AA:AA:AA:B0:D3:07:06:00:00:00:00:00:00:90:A2:E3:29:55:AA:AA:AA:B0:D2:07:06:00:00:00:00:00:00:D3:B6:98:3E:55:AA:AA:AA:B0:D1:07:06:00:00:00:00:00:00:16:8A:15:07:55:AA:AA:AA:B0:D3

  
</details>

As for the LEDs, the transistor can influence this. For example, because of the transistor, the LCD screen and lighting in the cup area flickered. Which transistor do you use? See a transistor that has a threshold gate voltage of +3v, for example a [BS170 transistor](https://pl-1.org/getproductfile.axd?id=9488&filename=BS170.pdf). 

[image]

What is the voltage at the output of the transistor? Should be +5V
[image]

DivanX10, 2023-05-26:
Could you share the log files as messages from the motherboard and from the control panel look like. Preferably a large log file. For example, do not click on the buttons and write to the log. I would like to compare them with my own. Are there functions B0/B5/BA/BB in the logs?

[image]

davidepanato, 2023-05-27:
Sorry I'm not at home in this weekend. I'll try to get logs next week. Anyway esp32 seems a good suggestion to me, I can try to give it a try

DivanX10, 2023-05-27:
> Sorry I'm not at home in this weekend. I'll try to get logs next week. Anyway esp32 seems a good suggestion to me, I can try to give it a try

Thanks, no need anymore. I just didn't understand how to disassemble the protocol for Philips 5400 and wanted to look at the protocol from Philips 2000 to compare and based on the author's documentation to understand where to look. But I got hooked on one thing, then another and it went and then I realized how to disassemble the protocol for Philips 5400 and disassembled it. I have published my work [here](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/issues/6#issuecomment-1565725971) and [here](https://community.home-assistant.io/t/esp-smart-offee-machine-philips-5400-series/568545/46?u=divanx10)

davidepanato, 2023-05-30:
Hello @DivanX10 and @TillFleisch I've just tried again today but I'm still facing some problems.
I switched to an esp32 device and mapped the part to 2 hw uarts.

here's my config:
```
uart:
  # UART connected to the mainboard
  - tx_pin: GPIO1
    rx_pin: GPIO3
    baud_rate: 115200
    stop_bits: 1
    data_bits: 8
    parity: NONE
    rx_buffer_size: 256
    id: uart_mainboard

        
  # UART connected to the display
  - tx_pin: GPIO17
    rx_pin: GPIO16
    baud_rate: 115200
    stop_bits: 1
    data_bits: 8
    parity: NONE
    rx_buffer_size: 256
    id: uart_display
```

Trnasisto GPIO has benn mapped to GPIO32.

Behavior is still the same, hw seems o because I can see communication and power working. I am able to power on the coffe machine via the display buttons, it detects some state change but then external display seems locked and nothing works.
Trying to power off the coffe machine from home assistant generates a couple of beep but it never shutdown.

Here's the state changes shown by home assistant at startup:
[image]

this is the communication that I can log (as suggested by @DivanX10 ):
`[21:57:14][D][uart_debug:114]: >>> 00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07
[21:57:14][D][uart_debug:114]: >>> 00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00
[21:57:15][D][uart_debug:114]: >>> 00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00
[21:57:15][D][uart_debug:114]: >>> 3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07
[21:57:15][D][uart_debug:114]: >>> 00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00
[21:57:15][D][uart_debug:114]: >>> 07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55
[21:57:15][D][uart_debug:114]: >>> 00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00
[21:57:15][D][uart_debug:114]: >>> 00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C
[21:57:15][D][uart_debug:114]: >>> 00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00
[... cut ...] 02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00`

Any idea on what is happening?

D5:55:00:00:00:03:02:00:00:00:2D:01 --> this seems to be a message from the display to the mainboard but it seems different compared to what you understood by reverse engineering the protocol.
Is it possible that I'm running a different sw on the coffe machine?

Thanks!
Davide

DivanX10, 2023-05-30:
Perhaps your switching on and off is different from the author's coffee machine. Try to catch the bytes responsible for switching on and off. This is usually seen at the beginning of the log when you turn on the coffee machine and almost at the end when you turn it off.  

For example, these bytes are responsible for my inclusion, but in fact I do not turn on when I send. Maybe you have the same, but to understand, first find these bytes. Before turning on the coffee machine, turn on the log entry, and then turn on the coffee machine, and then look for bytes in the logs

[image]

davidepanato, 2023-05-30:
I'll try that way but one point is still confusing me, I can accept that it is a different protocol but I do not understand why, after first messages, display is blocked and I cannot turn off the coffee machine anymore. If protocol is different I expect to not be able to decode it, not to block everything.

DivanX10, 2023-05-30:
> I'll try that way but one point is still confusing me, I can accept that it is a different protocol but I do not understand why, after first messages, display is blocked and I cannot turn off the [...]

As I assume that debagging is to blame for this, which has direction “BOTH” enabled by default. This affects the operation of the coffee machine, as data flies from the logs to the control and the coffee machine reads them, which causes failures. Personally, I came across this myself, I could not understand why the changed settings for making coffee were not saved, and it was all the fault of direction “BOTH” and TX, these two modes prevent the coffee machine from working normally when debagging
[image]

Specify that this is reading data and it will not affect the operation of the coffee machine, but at the same time it will be possible to control the coffee machine
```
   debug:
     direction: RX
```

I use this option

```
uart:
 - id: uart_display
   rx_pin: GPIO16
   tx_pin: GPIO17
   baud_rate: 115200
   stop_bits: 1
   data_bits: 8
   parity: NONE
   rx_buffer_size: 256
   debug:
     direction: RX
     dummy_receiver: false

 - id: uart_mainboard
   rx_pin: GPIO3
   tx_pin: GPIO1
   baud_rate: 115200
   stop_bits: 1
   data_bits: 8
   parity: NONE
   rx_buffer_size: 256
   debug:
     direction: RX
     dummy_receiver: false
```

TillFleisch, 2023-05-31:
`D5:55:00:00:00:03:02:00:00:00:2D:01` looks like a message coming from the display.
The log messages are a little confusing to me.

```
[21:57:14][D][uart_debug:114]: >>> 00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07:3F:38:00:00:00:00:07:00:00:00:00:34:1C:D5:55:00:07:07:07 [21:57:14][D][uart_debug:114]: >>> 00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00:00:00:03:02:00:00:00:2D:01:D5:55:00 [21:57:15][D][uart_debug:114]: >>> 
```
The first block holds messages from the motherboard to the display (longer messages). The second one holds messages which should arrive at the mainboard (shorter messages).
The direction indicator `>>>` implies that all messages are going in the same direction.

This may be the case if you set up UART debugging for both ports in the same direction. This could also be the case if the wiring is wrong.
Can you clarify how you gathered the logging data?
I would recommend using bi-directional logging on a single port like this:
```
    debug:
      direction: BOTH
      sequence:
        - lambda: UARTDebug::log_hex(direction, bytes, ' ');
```

Here is a short log snippet from starting my coffee machine:
``` 
[19:29:41][D][uart_debug:114]: >>> D5 55 0A 01 02 00 02 00 00 00 0E 12 D5 55 0A 01 02 00 02 00 00 00 0E 12 D5 55 0A 01 02 00 02 00 00 00 0E 12 D5 55 0A 01 02 00 02 00 00 00 0E 12 D5 55 0A 01 02 00 02 00 00 00 0E 12 D5 55 0A 01 02 00 02 00 00 00 0E 12 D5 55 02 01 02 00 02 00 00 00 38 15 D5 55 02 01 02 00 02 00 00 00 38 15 D5 55 02 01 02 00 02 00 00 00 38 15 D5 55 02 01 02 00 02 00 00 00 38 15 D5 55 02 01 02 00 02 00 00 00 38 15 D5 55 02 01 02 00 02 00 00 00 38 15
[19:29:41][D][uart_debug:114]: >>> 00
[19:29:41][D][uart_debug:114]: <<< D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00
[19:29:41][D][uart_debug:114]: >>> D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36
[19:29:41][D][uart_debug:114]: <<< 00 00 00 00 06 26 D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D D5 55 00 00 00 00 00 00 00 00 00 00 00
[19:29:41][D][uart_debug:114]: >>> D5 55 00 01 02 00 02 00 40 11 36 D5 55 00 01 02 00 02 00 00 40 44 F3 D5 55 00 01 02 00 02 00 00
[19:29:41][D][uart_debug:114]: <<< 00 00 00 00 39 0D D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D D5 55 00 00 00 00 00 00 00 00 00 00 00
[19:29:41][D][uart_debug:114]: >>> 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02
```
It looks like more content is buffered in your logs. Are you by any chance using the [dummy_receiver](https://esphome.io/components/uart.html#debugging)? If so disable it, as the component will not receive any data.

For further debugging, I would recommend the following steps:

1. Verify the machine is working as intended without the ESP installed.
2. To rule out specific problems with this component try using the external component  [uart_mitm](https://github.com/ssieb/custom_components/tree/master/components/uart_mitm). This performs the man-in-the-middle-attack used in my component without additional overhead (make sure the display has power by removing the transistor).

DivanX10, 2023-06-01:
@TillFleisch 
component [uart_mitm](https://github.com/ssieb/custom_components/tree/master/components/uart_mitm) I would not recommend it for the reason that because of it, the coffee machine turns off very quickly, after a few seconds. If you use your component, then the coffee machine works for a long time. For example, I set the standby mode for 180 minutes and if you use your component, the coffee machine will work for all 180 minutes and then go into standby mode, and in the case of uart_mitm, the coffee machine turns off in less than a minute. I don't understand why it's like this, apparently you did something in your component correctly, which allows the coffee machine to work correctly

davidepanato, 2023-06-11:
I think that in the end you guys were right :).
UART were working correctly my problem wast the transistor that was not correct for a 3v3 supply.
I switched to an IRL540N and now communication is working and I can send command while on.
What is still not working is the powering on procedure.
I measured the voltage and when transistor is opened it moves from 5V to 1.2V. Is this correct?
I'm worried that this is not enough to restart the display

Thanks!

DivanX10, 2023-06-12:
I may be wrong, but as I realized, using a transistor to turn on the coffee machine is not quite a working option and is not suitable for everyone. When I read the documentation of the chip built into the control panel, I found an interesting mention that you can control the coffee machine with bytes after the coffee machine is fully turned on, which means that you will not be able to turn on the coffee machine remotely with a command

Screenshot from the documentation on the chip in the coffee machine
[image]

By twisting the transistor, you thereby raise the voltage above 1.7 V and the coffee machine turns on, but this does not work for everyone. In your case, this may not happen for the reason that the coffee machine could have been released later than the author's and the manufacturer could have changed something, for example, your motherboard may differ from the motherboard of the author.

I realized turning the coffee machine on and off by soldering the wire to the resistor R110 and connected the relay single-channel 5V relay (KY-019). I do not know how my way of turning on and off the coffee machine can work on 1000/2000/3000 series coffee machines, I can only say about the 5000 series. You can disassemble your control panel, look at the board, track the tracks from the power button to the resistor and then it will become clear. I note that before deciding on such a step, I had previously tried to send a signal over the 4 wire, and then send a command to turn on the coffee machine, but this was not successful, so I took such measures to solder the wire to the resistor R110, since I did not see any other way.

In the photo, I drew all the tracks of the board using the touch buttons. We look at the red line from the power button. The red line leads to the resistor R110, to which I soldered the wire, thereby imitating pressing the touch button to turn on the coffee machine, but at the same time we lose the opportunity to turn on the coffee machine by touching with a finger. You can turn it on remotely in any way, through a voice assistant, through a zigby button, but turning it on with a touch of your finger no longer works. There is another way where it may work to turn on the coffee machine when touched and turn on remotely, but I did not try this method because I was afraid that I might damage the chip in the control board or something else. I tested this option on a table lamp with a touch button and this option is fully working, but I do not dare to use a coffee machine. I'm not ready for such experiments. In order for the button to work at the touch of a finger and the coffee machine to turn on remotely, you need to solder the wire after the resistor R110, and not before the resistor R110. The resistor removes static and interference, and if we solder after the resistor, then we will send a parasitic capacitance directly to the chip and I do not know how the chip will react, so I'd rather turn it on remotely than risk it.

[image]
[image]
[image]

Connection diagram of the relay to the control panel. Don't pay attention to the servo, I made the water drain for myself. For understanding, you can watch my [video1](https://drive.google.com/file/d/1ukaCe0znJzEZIeUdLH1dEzmpiDbQ_PVg/view?usp=drive_link) and [video2](https://drive.google.com/file/d/1qglbbCl32KrK460X2jpunkvSGTnTsLG6/view?usp=drive_link)
[image]

Installed the relay on the back of the control panel
[image]

Add a hidden switch and a button to turn the coffee machine on and off

```
switch:
  - platform: gpio
    pin: GPIO14
    name: "Relay Switch"
    id: idRelaySwitch
    internal: True
    restore_mode: ALWAYS_OFF

button:
#Turn on the coffee machine
  - platform: template
    name: "Power"
    icon: mdi:power
    on_press:
     - switch.turn_on: idRelaySwitch
     - delay: 1s
     - switch.turn_off: idRelaySwitch
```

TillFleisch, 2023-06-13:
The brown out information @DivanX10 has posted seems very useful. As mentioned this may be different on the EP2200.

1.2 volts may not be low enough, but we don't know if this is the case for sure.
You could try to perform the power on trick manually by physically disconnecting and re-connecting the wires (using a button) to check if this the power trick is possible on your revision of the coffee machine. Make sure to disconnect both ground wires as shown in the diagram. Before/After reconnecting the power on message must be sent to the Mainboard, this can be achieved by using the power switch as usual.

Additionally you could also try to modify the time for which the output pin is changed (see [power_switch](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/blob/main/components/philips_power_switch/power.h)).
In case there are some capacitors which don't drain fast enough, increasing the power trip delay value may be helpful. The half a second delay should be long enough though. I wouldn't put a lot of hope in this approach.

DivanX10, 2023-06-13:
> The brown out information @DivanX10 has posted seems very useful. As mentioned this may be different on the EP2200.
>
> 1.2 volts may not be low enough, but we don't know if this is the case for sure. [...]

Correct me if I didn't understand you correctly from what I read. How did I understand? We disconnect the two wires + and GND and connect them to the relay, then through the switch we realize the voltage supply to the control panel and immediately send a command to turn on. The principle is similar to a light bulb when the switch is turned on and the light bulb lights up. If this is the case, then I doubt it will work, since this method is not much different from the method when I soldered the transistor, where I opened and closed the GND, thereby supplying power to the control panel. If the plug is removed from the outlet and the plug is inserted into the outlet again, the coffee machine does not respond to this. I note that if the screen would turn on when the voltage was applied to the control panel, then it was enough to remove the plug from the socket and plug it back into the socket. After all, plugging the plug into the socket, there is a voltage of +4.95 v in the + and GND wires. I saw in the logs that bytes ran after opening and closing, but then they stopped. I was sending commands at that moment and I tried different options. It was to no avail

closed 2024-01-07 by TillFleisch