TillFleisch/ESPHome-Philips-Smart-Coffee #11: Protocol Decoding

adn77, 2023-05-30:
This is not really an issue, but rather a kind request for further input...

I have been reading up on most of the available documentation of the serial protocol.

My ultimate goal is to send a sequence for a recipe which is not available on the display of my EP223x Latte go.

1. In order to try other values for the data bytes, one would have to find a way to do the checksum calculation. Although it would seem a waste of resources to use a 16bit checksum on this short payload, I went ahead with that very same assumption - and didn't get anywhere. I googled for 16 bit checksums but only came to https://crccalc.com. It might still be a custom CRC with special init parameters... but why make things so complicated.
3. Following the recent adventures of [@DivanX10](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/issues/6)  I was asking myself if the EP5xxx protocol would also work on the smaller EP22xx machines. That is under the assumption that the mainboard is the same for all products (as stated somewhere) running the same firmware.

I have yet to plugin in my MITM board. ESP and cables are just waiting to be soldered together. So please forgive me for these theoretic ramblings...

Any further insights are highly appreciated!

TillFleisch, 2023-05-30:
I had a look at https://crccalc.com/ previously and came to the same conclusion as you. I have no idea why they are using a 16 bit checksum, but my standing desk does so as well and it's using even shorter payloads. Maybe this is common practice or the result of some lower level UART hardware implementation.

I would love to know how the checksum is calculated. This would make injecting messages easier.

I doubt that the commands from the ep5xxx work on the ep22xx.
The handbook/documentation of the coffee machine allows the user to change certain parameters (like brew temperature). Maybe this is helpful for you. This could in theory also be automated sice we are just simulating user input.

DivanX10, 2023-05-30:
I partially deciphered the protocol of coffee drinks for Philips 5400 and made a table to clearly see and understand what and how, and also described in detail how I got the [recipes](https://community.home-assistant.io/t/esp-smart-offee-machine-philips-5400-series/568545/49?u=divanx10). Perhaps my experience will help you understand how to decipher your coffee drink protocol for EP223x Latte go.

Decoding of the protocol of coffee drinks for Philips 5400
[image]

TillFleisch, 2023-07-03:
@adn77 can we close this issue?

closed 2024-01-07 by TillFleisch