TillFleisch/ESPHome-Philips-Smart-Coffee #12: Sensor for Bean/Coffee Size

Joshi425, 2023-07-01:
Hello,

nice work with this component, works like a Charm 👌 
I was wondering if there is an easy way to access the currently selected Bean/Coffee Size. At the moment I only see the Text Status Sensor.

When the mosfet resets the display Unit on my Maschine sometimes the settings for Bean/Coffee Size reset to 2/2. For A fully automated 
Coffee it would be great to have those as a sensor to make the Coffee like I want. It looks like its also already decoded 

https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/blob/main/protocol.md#messages-from-the-display-to-the-mainboard-1

so should be trivial to implemet right?

thanks for your great Work.

Joshi

TillFleisch, 2023-07-02:
@Joshi425 can check out/give feedback on the new feature by adding the following to your configuration:

```yaml
external_components:
  - source: github://TillFleisch/ESPHome-Philips-Smart-Coffee@bean_and_water_sensor

number:
  - platform: philips_bean_settings
    name: "Coffee beans"
    controller_id: philip
    status_sensor_id: status
    source: COFFEE

  - platform: philips_size_settings
    name: "Coffee size"
    controller_id: philip
    status_sensor_id: status
    source: COFFEE
```

TillFleisch, 2023-07-08:
>instead of the Steam button I have a Cappuccino Button

Well that makes adding this as an option very easy. I've updated the PR.

You can now add `use_cappuccino = true` to the status sensor and it will report "Cappuccino selected" instead of "Steam selected".
The water and size settings now also have a `CAPPUCCINO` option.

>It only works for Cofee, not for Hot Water, Cappuccino or Espresso

I've noticed that the automatic selection sometimes overshoots the selected value. I have reduced the speed at which the buttons are pressed, this should fix at least this issue.

Does the status sensor on your machine report the correct status? (like `Hot water selected`)
If you change the the selected size/bean count manually for Hot Water, Cappuccino or Espresso, is the correct value reported?

Joshi425, 2023-07-10:
thanks, now it shows `Cappuccino selected` 👍 

more delay feels better, I've noticed those overshoots as well but since it still worked it wasn't that bad. 

now if something else then Coffee is selected it shows the bean/coffee size numbers as unknown and its not possible to set them. it works for coffee tho

TillFleisch, 2023-07-10:
Did you set the desired source for the size/bean settings?

Quote from readme:
> - **source**(**Required**, int): The source of this sensor. Select one of `COFFEE`, `COFFEE`, `CAPPUCCINO`, `HOT_WATER`. When selecting `CAPPUCCINO` the related status sensor must use `use_cappuccino = true`.

The coffee number selection sensor from the example configuration above will only work when coffee is actively selected.
Can you verify that the settings work for cappuccino by adding a size/bean number using the `CAPPUCCINO`  source?

Joshi425, 2023-07-14:
aye, yeah didn't notice that, I've added size and bean size for every type, it works perfectly also for CAPPUCCINO 👍 the other numbers switch to unknown when something else is selected. But I think thats fine. 

in the Readme there is currently a small ISSUE, it says COFFEE twice instead of ESPRESSO on the 2 size sensors.

TillFleisch, 2023-07-14:
>  the other numbers switch to unknown when something else is selected.

This is expected behavior. 

> in the Readme there is currently a small ISSUE, it says COFFEE twice instead of ESPRESSO on the 2 size sensors.

Thank you for pointing this out.

closed 2023-07-14 by TillFleisch

TillFleisch, 2024-02-05:
@Joshi425, I was looking into different versions of the coffee machine. Based on the information provided in this issue I guess you're using the `EP2235`. Can you confirm this?