TillFleisch/ESPHome-Philips-Smart-Coffee #14: Error in configuring status sensor

davidepanato, 2023-07-20:
Hello Till,
I've tried to enable new config but it is throwing an error when nbuilding:
How should I configure the status sensors?

Thanks,
Davide

number.philips_bean_settings: [source /config/philips-ep2200.yaml:124]
  platform: philips_bean_settings
  name: Coffee beans
  controller_id: philip
  
  Couldn't find ID 'status'. Please check you have defined an ID with that name in your configuration.
  status_sensor_id: status
  source: COFFEE
  disabled_by_default: False
  mode: SLIDER
number.philips_size_settings: [source /config/philips-ep2200.yaml:130]
  platform: philips_size_settings
  name: Coffee size
  controller_id: philip
  
  Couldn't find ID 'status'. Please check you have defined an ID with that name in your configuration.
  status_sensor_id: status
  source: COFFEE
  disabled_by_default: False
  mode: SLIDER

TillFleisch, 2023-07-21:
Does your `philips_status_sensor` have the id `status`?
The `status_sensor_id` on bean/size settings must be set to the id of the status sensor in your configuration.

You can add a status sensor with the id `status` like this:

```
text_sensor:
  - platform: philips_status_sensor
    controller_id: philip
    id: status
    name: "Status"
```

davidepanato, 2023-07-21:
Thanks a lot, I was missing the id filed!

closed 2023-07-21 by davidepanato