TillFleisch/ESPHome-Philips-Smart-Coffee #15: Which parts do we need?

djacid, 2023-11-14:
Hello, first of all congrats for this repository. 
Is it possible to tell me which parts I need to replicate this? As I can see I need a mosftet and a cable with picoflex. Is it possible to update your guide with reference of these parts? Thank you in advance

TillFleisch, 2023-11-14:
I used a Wemos D1 Mini and a IRLZ44N Mosfet (which is probably overkill), as I had these parts laying around.
You can use pretty much any ESP2866 or ESP32 but make sure to use a voltage regulator to lower the 5V provided by the machine to the 3.3V required by the ESPs. Otherwise you will end up releasing the magic smoke.
As for the mosfet/transistor, you can use pretty much anything that is capable of switching 5V on/off. Please keep in mind that I have no background in electrical engineering. Proceed with caution and at your own risk.

djacid, 2023-11-14:
Thank you for your response. I am talking about the cables and the connectors. Could you please give me more details about them?

djacid, 2023-11-14:
Is this cable suitable?
[image]

TillFleisch, 2023-11-14:
You can either cut the existing ribbon cable in half (which will definitely void your warranty) or get some male and female 8 pin picoflex connectors. The one in the image looks fine, but you will also need the opposing connector. Maybe there are 8-pin extension cables, which already have both necessary connectors. I would recommend opening the front panel on your machine and familiarizing yourself with the connectors.

djacid, 2023-11-19:
Is it possible to tell me where could I find the correct pic oflex connector?

sepodele, 2023-11-20:
The 8P connector here: https://www.aliexpress.com/item/1005005142277556.html
The 8P 'Same direction' cable here: https://www.aliexpress.com/item/1005005026658462.html

bkbartk, 2023-11-22:
> The 8P connector here: https://www.aliexpress.com/item/1005005142277556.html The 8P 'Same direction' cable here: https://www.aliexpress.com/item/1005005026658462.html

thnx, 
I just ordered those,
I think this is better then what I have already in backorder,

sepodele, 2023-12-08:
Just soldered onto the connector.

bkbartk, 2023-12-08:
I was afraid you would say that,
my soldering skills are not that good,
I was already happy I finished soldering the wemos
I will give the a try some other time,
In the meanwhile I ordered these
https://www.aliexpress.com/item/1005002804645942.html?spm=a2g0o.detail.0.0.529bFc0LFc0LeH&mp=1
in case soldering fails

djacid, 2024-01-26:
Is it the same if I use an IRFZ44N and not an IRLZ44N?

TillFleisch, 2024-01-26:
The IRFZ44N has a higher gate threshold voltage and therefore the ESP may not be able to toggle the mosfet fully. As a result the display power trip may not work. Your can find more details in #8. You may be able to verify wether your ESP is able to trigger the mosfet by measuring the output voltage.

TillFleisch, 2024-02-10:
[See here](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/issues/30#issuecomment-1937103408). Looks like everything is running.

closed 2024-02-10 by TillFleisch