TillFleisch/ESPHome-Philips-Smart-Coffee #18: question: is there a reason why all gounds are connected?

bkbartk, 2023-12-22:
Hello,

Is there a reason why all Grounds on pins 1 and 2 are connected?
I only connected the ground on pin 1 on both sides.

This gives some strange behaviar, the esp led is turned on, but esp in not accessible through webinterface or HA. that's the part I think is strange,

And no leds on the display of the coffee machine are turned on at any point of time, even not after removing the mosfet.
If this would be the reason I need both grounds I will change it(probabaly after the holidays)

but I actually think there is some other reason stuff is not working.

TillFleisch, 2023-12-22:
I measured continuity on pins 1 and 2 and wired them together to change as few things as possible. If we assume they are connected internally on both sides it should make no difference (I have never tried this).
The ESP not starting may be a bootstrapping-pin issue. Also make sure that baud rate on the logger component is set to 0 to prevent ESPHome from writing to the bus.

bkbartk, 2023-12-22:
`If we assume they are connected internally on both sides it should make no difference`
that was indeed my assumption, I hav never seen this to be different, but you never know.
`The ESP not starting may be a bootstrapping-pin issue.`
I think you might be correct here.

I get this warning on build
`WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.`
I didn't think much of it, but since I read about it, I should avoid.
I use a wemos D1 esp32, where GPIO5 is located on the position of GPIO15 of the original wemos.
So probably I need another pin for that, 
Do you perhaps know if it matters which pin I select? before reading this project I thougth rx/tx was only possible on GPOI 1 and 3.

`Also make sure that baud rate on the logger component is set to 0 to prevent ESPHome from writing to the bus.`
that one I copied from the example, so that should be ok. only just realized that I could have tried serial logging and then found that it wasn't working either.

TillFleisch, 2023-12-22:
If you are using an ESP32, I would recommend using the hardware serial pins for the two UARTs. If your using other pins, ESPHome will use a software serial implementation (which is usually a tad slower).
As for the mosfet-pin, you can use any pin which can be used as an output.

bkbartk, 2023-12-23:
thnx for the suggenstion for the uarts,
I didn't know, but I will use them,
I think I need to connect both GND. as I tried only connecting 5v and GND on pin 1 and the esp still wouldn't boot as expected.

You previously mentioned
`to change as few things as possible`
and I now read the docs carefully and see this
`The unlabeled wires should be connected without changes.`
which I missed, So I just ignored them completely

So I also have some additional soldering to do, I hope I can still to this before the holiday's

closed 2024-03-16 by TillFleisch