TillFleisch/ESPHome-Philips-Smart-Coffee #20: component formatting

poldim, 2024-01-15:
I tried to get some help for customizing this component and *ssieb* on discord mentioned that the formatting wasn't the best.  Since I have very little clue of what I'm doing in coding, I can't exactly make out how to modify my clone of your code and have it compile.  However, I thought there's a chance that you might want to know this and modify your code.  Keep in mind, the example I sent him was probably from ~6-12 months ago when I cloned your repo and modified it to work with my 3200.

[image]

His repo: https://github.com/ssieb/esphome_components/tree/espresso/components/philips_espresso

https://discord.com/channels/429907082951524364/1168697647276769390/1169195798014214165

poldim, 2024-01-15:
BTW, this is what I used to copy that repo but I ultimately got lost trying to make the changes:
`git clone --branch espresso --single-branch https://github.com/ssieb/esphome_components/`

TillFleisch, 2024-01-16:
I'm sorry, I don't  understand what this issue is about. Do you need help with your fork of this component or is this just a heads-up regarding the file/project structure of this external component (as pointed out by jesse on discord).
After reading through the linked discord thread, I see how there if room for improvement regarding the project structure. I have opend a separate issue(#21) for refactoring but I can't promise anything anytime soon, as I'm currently quite busy.