TillFleisch/ESPHome-Philips-Smart-Coffee #22: Ground coffee selection

mikulik86, 2024-01-23:
As there is this function missing, I tried to implement it simply in YAML using uart.write action. Because a 3s long press is required, I set the uart.write action to be repeated 70x. This approach kind of works, the ground coffee is selected, but it causes the esp8266 to reboot.
Could you please help me to troubleshoot it?

I know that it is not really an Issue of this component, but as there is support for AQUA_CLEAN button which doesn't work (because there is not implemented the 3s long press, i suspect), so I think it is kind of relevant.

TillFleisch, 2024-01-23:
#23 adds support for button long presses. You can try this in your configuration by using:
```
external_components:
  - source: github://TillFleisch/ESPHome-Philips-Smart-Coffee@button-long-press-support
```

Can you verify that this is solving your issue. It's working on my machine, I jsut want to double check.

mikulik86, 2024-01-25:
Works like a charm! Thank you.

PR #23 (2024-01-25): Add long press to action button
This PR adds support for long button long presses to the action button component.

Close #22
closed 2024-01-25 by mikulik86