TillFleisch/ESPHome-Philips-Smart-Coffee #30: Turning on machine issue

djacid, 2024-02-02:
First of all thank you very much for your responsiveness.
After all your answers everything is working almost perfect.
I have only one more issue. 
I can turn on the machine and manage it without any issues via screen.
The only problem that I have is whenever I want to turn on the machine with esp. It needs to be pressed fastly 2-3 times to succeed. I read that this issue could be related to mosfet but I used an IRLZ44N. So do you have any idea what it could be?

TillFleisch, 2024-02-02:
> whenever I want to turn on the machine with esp. It needs to be pressed fastly 2-3 times to succeed.
Here are some trouble shooting steps you can take.
- double check your wiring and ensure grounds are connected properly
- try de/increasing the `power_trip_delay` option on the `philips_series_2200` hub
- try increasing [`MAX_POWER_TRIP_COUNT`](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/blob/5ec5f6c1fb3ea7edb9d533efab369011ba0b9350/components/philips_power_switch/power.h#L9C1-L9C31) (requires modifying source)

I *think*, increasing the `power_trip_delay` will solve this issue, but you never know.

djacid, 2024-02-03:
I checked all the grounds and they doesn’t seem to have issues.
I increased power trip delay and it didn’t work.
So, I will try last option and I will post again later

djacid, 2024-02-03:
Finally, I found the problem. I read the closed issues on Github and found someone who had the same problem. By increasing MESSAGE_REPETITIONS to 24 everything is working perfect.
It would be nice, if you could add a new option to the esp yaml configuration file for message repetitions.

TillFleisch, 2024-02-03:
#19 for reference.

This is wierd behaviour. I will look into how this can be exposed to config/fixed. I guess every machine/esp is just built a little different...

TillFleisch, 2024-02-03:
I will observe how this change affects my machine, as it has worked previously, to ensure no wierd behaviour was introduced. This change should solve your problem, if so we can close this issue once merged.

In the meantime you can try the change by adding 
```yaml
external_components:
  - source: github://TillFleisch/ESPHome-Philips-Smart-Coffee@increase_max_repetitions_fix_power_on
```
to your config.

djacid, 2024-02-04:
It's not working with this source. I got this error:
"Component number.philips_bean_settings requires component philips_status_sensor."

TillFleisch, 2024-02-04:
You are still using a v1 yaml configuration, see [v2.0.0 Release Notes](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/releases/tag/v2.0.0) which include breaking changes.

djacid, 2024-02-04:
I updated my configuration to V2. I also changed the source url as you told me. I have another issue now.
If I turn on the machine, make coffee and press again to turn off the machine, it seems that the only thing that turns off is the display and not the machine. Ηow do i know this? If I press again to turn on the machine, machine makes a sound and display stays still off.

TillFleisch, 2024-02-04:
> If I press again to turn on the machine

Do you mean the physical button on the display or the one provided by this component?

> it seems that the only thing that turns off is the display and not the machine

Does it make a beep sound when turning off (through the component)? What does the status sensor report? A beep and the display not turning on may just be the same problem as we've started with and the PR doesn't change anything.

This component sends a message to the `mainboard` to turn off, which then intern does some magic and tells the display to turn off/stops interacting with the display. While it's probably possible to spoof the display into turning off, I don't see how this could happen with the current state of this external component as all messages from the mainboard to the display are forwarded and not altered.

djacid, 2024-02-04:
_Do you mean the physical button on the display or the one provided by this component?_
No, I mean this one provided by component.

_Does it make a beep sound when turning off (through the component)? What does the status sensor report? A beep and the display not turning on may just be the same problem as we've started with and the PR doesn't change anything._

Yes it does. There is beep sound as it should be. The status sensor doesn't seem to be reliable. Machine is on but the sensor says off. Basically I think something else is happening. When I press (component) to turn on, the machine does not make any beep sound. Only the screen just turns on. After that, if I press (component) to turn off, the machine and the display turns off with beep sound. Then, if I press again to turn on, you can easily listen the beep sound along with the sound of a machine's mechanism as it should be, but the display stays off. Probably the machine is on but the status sensor says off. After that, if I press again to turn on, the display turns on without beep sound.

TillFleisch, 2024-02-04:
Hm... the change in #34 is slightly different from your solution. I increased the repetition count to `50` (to be on the safe side) only for power-related messages, not all messages. I have now lowered this to `25` (the assumption being that the very high repetition count causes some interference with other messages from the display). I will investigate the original power-off behaviour (messages on the bus) later once more. Maybe I have overlooked something.

djacid, 2024-02-04:
Even though you changed the repetition count back to 25, the issue remains. I really don't get it :(

djacid, 2024-02-04:
For one more time, you had right! That was the problem. Now it works as before, without issues.

TillFleisch, 2024-02-04:
I still don't understand _why_ the high message repetition count is a problem (the machine _should_ just discard/handle duplicate messages as the display also generates these - it works on my machine), but I'm happy it works. Let's keep an eye on this and see if this is consistent behaviour.

djacid, 2024-02-04:
I didn't have refresh option on my configuration. Maybe that was the reason.

TillFleisch, 2024-02-09:
@djacid: Everything is running as usual on my machine. Is everything consistently working as expected on your setup?

djacid, 2024-02-10:
Yes everything is working flawlessly. I use the branch "increase_max_repetitions_fix_power_on"

closed 2024-02-10 by TillFleisch

TillFleisch, 2024-02-14:
#39 returns the default repetition value back to `5` and provides configuration of this value via the YAML configuration.

PR #34 (2024-02-10): Fix unreliable power-on on some machines
This PR increases the message repetition count for power related messages. As mentioned in #19 and #30 this resolve a `powering on` issue on some machines. This change should have no side effects on other machines.

Close #30
Close #19