TillFleisch/ESPHome-Philips-Smart-Coffee #36: Philips EP3221 Support

machintrucbidule, 2024-02-07:
Hi,

As discussed in the EP3243 thread, I open a new one for 3221 support.
Indeed if both are part of the 3200 series, they have different options and code shouldn't be fully compatible.

EP3221 dashboard looks like this:
[image]

On left coffee part I have: Expresso, Expresso lungo, Coffee, Americano, Hot Water, Steam.
And on the right part: Bean, Size, Temperature, and then Aqua Clean and Clean.

I ordered hardware to build this (still waiting for it), but would you be able to tell me how to adapt the code for my machine? How to identify which command is sent when pressing a button, and what should I update in the code in order to add compatibility with my 3221 while not breaking 3243 one?

Thanks!

TillFleisch, 2024-02-07:
> How to identify which command is sent when pressing a button

You can use the ESPHome UART debug functionality to sniff what is happening on the bus. Based on that information you can infer codes for button presses and display messages.

I guess the `EP3221` codes will be very similar / the same as the `EP3242` commands, since they are both 3200 series. Apart from the Americano/Cappuccino option the LED-layouts are the same, indicating that the command are also probably the same. The `EP2220` commands are also compatible with the `EP2235`, but the `EP2235` has a Cappuccino option instead of Steam.

If the commands are the same, it's probably a just a matter of re-naming things. I would recommend waiting until #35 is merged (or #7 is closed through a different PR) and using the `EP3242`-related changes as a guide. Additionally, I would recommend creating a protocol overview for messages from/to the mainboard as shown [here](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/blob/main/protocol.md) and [here](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/issues/7#issuecomment-1932889858).

machintrucbidule, 2024-02-07:
Ok, thanks, I'll check the UART debugging when I'll have receive and installed the hardware.
And there's also another difference than Amecerican/Cappuccino which is the far right option. On 3243 that's to change milk settings, on 3221 for temperature selection. But here too I guess that could be just a matter of renaming ?

TillFleisch, 2024-02-24:
Comment from [here](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/issues/40#issue-2151988602) (#40)
> Hi,
> 
> I have an issue when trying to turn on my coffee machine thru HA. When clicking on the Power switch, the machine beeps, turns on; but the switch goes back to Off, and display remains off.
> 
> Logs:
> 
> > [01:17:10][D][switch:012]: 'Power' Turning ON.
> > [01:17:10][D][uart_debug:114]: >>> D5:55:0A:01:03:00:0E:00:00:00:2A:10
> > [01:17:10][D][uart_debug:114]: >>> D5:55:0A:01:03:00:0E:00:00:00:2A:10
> > [01:17:10][D][uart_debug:114]: >>> D5:55:0A:01:03:00:0E:00:00:00:2A:10
> > [01:17:10][D][uart_debug:114]: >>> D5:55:0A:01:03:00:0E:00:00:00:2A:10
> > [01:17:10][D][uart_debug:114]: >>> D5:55:0A:01:03:00:0E:00:00:00:2A:10
> > [01:17:10][D][uart_debug:114]: >>> D5:55:0A:01:03:00:0E:00:00:00:2A:10
> > [01:17:10][D][uart_debug:114]: >>> D5:55:01:01:03:00:0E:00:00:00:01:25
> > [01:17:10][D][uart_debug:114]: >>> D5:55:01:01:03:00:0E:00:00:00:01:25
> > [01:17:10][D][uart_debug:114]: >>> D5:55:01:01:03:00:0E:00:00:00:01:25
> > [01:17:10][D][uart_debug:114]: >>> D5:55:01:01:03:00:0E:00:00:00:01:25
> > [01:17:10][D][uart_debug:114]: >>> D5:55:01:01:03:00:0E:00:00:00:01:25
> > [01:17:10][D][uart_debug:114]: >>> D5:55:01:01:03:00:0E:00:00:00:01:25
> > [01:17:10][W][component:214]: Component api took a long time for an operation (0.17 s).
> > [01:17:10][W][component:215]: Components should block for at most 20-30ms.
> > [01:17:10][D][uart_debug:114]: <<< D5:55:01:00:00:00:00:00:00:00:00:00
> > [01:17:10][D][uart_debug:114]: <<< 00:00:00:00:00:06:26:D5:55:01:00:00
> > [01:17:10][D][uart_debug:114]: >>> D5:55:01:00:00:00:00:00:00:00:00:00
> > [01:17:10][D][uart_debug:114]: >>> 00:00:00:00:00:06:26:D5:55:01:00:00
> > [01:17:10][D][number:012]: 'Coffee beans': Sending state nan
> > [01:17:10][D][number:012]: 'Coffee size': Sending state nan
> > [01:17:10][W][component:214]: Component esphome.coroutine took a long time for an operation (0.06 s).
> > [01:17:10][W][component:215]: Components should block for at most 20-30ms.
> > [01:17:11][W][component:214]: Component esphome.coroutine took a long time for an operation (0.75 s).
> > [01:17:11][W][component:215]: Components should block for at most 20-30ms.
> > [01:17:11][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00
> > [01:17:11][D][uart_debug:114]: >>> 00:00:00:00:00:00:00:00
> > [01:17:11][D][uart_debug:114]: >>> 00
> > [01:17:11][D][uart_debug:114]: <<< 00
> > [01:17:11][D][uart_debug:114]: <<< 00:00:00:00:06:26:D5:55:01:00:00:00
> > [01:17:11][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:06
> > [01:17:11][D][uart_debug:114]: <<< 26:D5:55:01:00:00:00:00:00:00:00:00
> > [01:17:11][D][uart_debug:114]: >>> 00:00:00:00:06:26:D5:55:01:00:00:00
> > [01:17:11][D][uart_debug:114]: >>> 00:00:00:00:00:00:00:00:00:00:00:06
> > [01:17:11][D][uart_debug:114]: >>> 26:D5:55:01:00:00:00:00:00:00:00:00
> > [01:17:11][D][philips_power_switch:089]: Performed 1 power trip(s).
> > [01:17:11][D][switch:055]: 'Power': Sending state ON
> > [01:17:11][W][component:214]: Component esphome.coroutine took a long time for an operation (0.12 s).
> > [01:17:11][W][component:215]: Components should block for at most 20-30ms.
> > [01:17:11][D][uart_debug:114]: >>> 00:00
> > [01:17:11][D][uart_debug:114]: <<< D5:55:00:01:04:02:04:00:00:00:2F:0D
> > [01:17:11][D][uart_debug:114]: <<< D5:55:00:01:04:02:04:00:00:00:2F:0D
> > [01:17:11][D][uart_debug:114]: <<< 00:00
> > [01:17:11][D][uart_debug:114]: >>> D5:55:00:01:04:02:04:00:00:00:2F:0D
> > [01:17:11][D][uart_debug:114]: >>> D5:55:00:01:04:02:04:00:00:00:2F:0D
> > [01:17:11][D][uart_debug:114]: >>> D5:55:00:01:04:02:04:00
> > [01:17:11][D][uart_debug:114]: <<< D5:55:00:01:04:02:04:00
> > [01:17:11][D][uart_debug:114]: <<< 00:00:00:00:06:26:D5:55:01:00:00:00
> > [01:17:11][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:06
> > [01:17:11][D][uart_debug:114]: <<< 26:D5:55:01:00:00:00:00:00:00:00:00
> > [01:17:11][D][uart_debug:114]: >>> 00:00:00:00:06:26:D5:55:01:00:00:00
> > [01:17:11][D][uart_debug:114]: >>> 00:00:00:00:00:00:00:00:00:00:00:06
> > [01:17:12][D][uart_debug:114]: >>> 26:D5:55:01:00:00:00:00:00:00:00:00
> > [01:17:12][W][component:214]: Component esphome.coroutine took a long time for an operation (0.19 s).
> > [01:17:12][W][component:215]: Components should block for at most 20-30ms.
> > [01:17:12][D][uart_debug:114]: >>> 00:00
> > [01:17:12][D][uart_debug:114]: <<< 00:00:2F:0D:D5:55:00:01:04:02:04:00
> > [01:17:12][D][uart_debug:114]: <<< 00:00:2F:0D:D5:55:00:01:04:02:04:00
> > [01:17:12][D][uart_debug:114]: <<< 00:00
> > [01:17:12][D][uart_debug:114]: >>> 00:00:2F:0D:D5:55:00:01:04:02:04:00
> > [01:17:12][D][uart_debug:114]: >>> 00:00:2F:0D:D5:55:00:01:04:02:04:00
> > [01:17:12][D][uart_debug:114]: >>> 00:00:2F:0D:D5:55:00:01
> > [01:17:12][D][uart_debug:114]: <<< 00:00:2F:0D:D5:55:00:01
[... 250 lines cut ...]
>     source: COFFEE
> 
>   - platform: philips_coffee_machine
>     type: size
>     name: "Coffee size"
>     controller_id: philip
>     status_sensor_id: status
>     source: COFFEE
> ```

Can you generate a log only containing uart logging for the display side?
```yaml
    id: uart_display
    debug:
      direction: BOTH
      sequence:
        - lambda: UARTDebug::log_hex(direction, bytes, ' ');
```
From the logs it is not clear where messages are originating from.

>[01:17:11][D][philips_power_switch:089]: Performed 1 power trip(s).
>[01:17:11][D][switch:055]: 'Power': Sending state ON

It looks like the display has sent at least one ~~valid~~ message, therefore this component assumes the display is turned on.
Maybe some basic message validation is required [for this operation](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/blob/00389f9e111e4602cc05810421ec3ecb19b60abb/components/philips_series_2200/philips_series_2200.cpp#L46) to prevent garbled messages/zeros from setting the flag.

machintrucbidule, 2024-02-24:
Hi,

Thanks for your support!

And weird, when I wanted to check this, the ESP was offline, even if coffee machine was still powered (I have been able to turn it on with the display). If the ESP crashed, is there any other way than letting the wireless debug activated to know why?

For the display logs only:

> [11:53:07][D][switch:012]: 'Power' Turning ON.
> [11:53:08][W][component:214]: Component esphome.coroutine took a long time for an operation (0.75 s).
> [11:53:08][W][component:215]: Components should block for at most 20-30ms.
> [11:53:08][D][philips_power_switch:089]: Performed 1 power trip(s).
> [11:53:08][D][switch:055]: 'Power': Sending state ON
> [11:53:08][D][uart_debug:114]: <<< 00 D5 55 00 01 04 02 04 00 00 00 2F 0D D5 55 00 01 04 02 04 00 00 00 2F 0D D5 55 00 01 04 02 04 00 00 00 2F 0D
> [11:53:08][D][number:012]: 'Coffee beans': Sending state nan
> [11:53:08][D][number:012]: 'Coffee size': Sending state nan
> [11:53:08][D][uart_debug:114]: >>> D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26
> [11:53:08][D][uart_debug:114]: <<< D5 55 00 01 04 02 04 00 00 00 2F 0D D5 55 00 01 04 02 04 00 00 00 2F 0D D5 55 00 01 04 02 04 00
> [11:53:08][D][uart_debug:114]: >>> D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26
> [11:53:08][D][uart_debug:114]: <<< 00 00 2F 0D D5 55 00 01 04 02 04 00 00 00 2F 0D D5 55 00 01 04 02 04 00 00 00 2F 0D D5 55 00 01
> [11:53:08][D][uart_debug:114]: >>> D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26
> [11:53:08][D][uart_debug:114]: <<< 04 02 04 00 00 00 2F 0D D5 55 00 01 04 02 04 00 00 00 2F 0D D5 55 00 01 04 02 04 00 00 00 2F 0D
> [11:53:08][D][uart_debug:114]: >>> D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00
> [11:53:08][D][uart_debug:114]: <<< D5 55 00 01 04 02 04 00 00 00 2F 0D
> [11:53:08][D][uart_debug:114]: >>> 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26
> [11:53:08][D][switch:055]: 'Power': Sending state OFF

But know the machine doesn't turn on again.
I checked with previous logging for both mainboard and display, machine turns on. But when only logging display, that's no more the case ?!

TillFleisch, 2024-02-24:
> is there any other way than letting the wireless debug activated to know why?

You *could* enable logging via the uart ports. It would interfere with communication on the bus. Using a different set of pins  (for one uart) would be required. This may be possible using software-serial (depending on your board), which I have not played around with on the EP2220. Be also aware of power related things. Power/GND supplied by the coffee machine may do wierd things to your computer. I've heard from someone breaking their USB port/pc this way. I am no expert when it comes to electronics/power-related things.

> I checked with previous logging for both mainboard and display, machine turns on. But when only logging display, that's no more the case ?!

Enabling uart logging slows the ESP down a little, as there is more to do. Therefore timings may be different depending on the configuration. Disabling uart logging may improve/change things.

The custom component should try to perfom the power-trip multiple times (should be visible in the logs) which is similar to turning on the power-switch multiple times. Based on your logs it looks like the display starts sending messages, indicating it's on (therefore only a single power trip was performed in this case).
This behviour (display not lit but sending messages) is new to me. Does it react to button presses other than the power button in this state?

What's also interesting is that the mainboard is always sending `D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26`.
If I recall correctly this messages tells the display to turn off? The third byte (in this case `01`) is usually `00` for normal status update messages. After turning on some LEDs are usually lit, in this case they are all intended to be off (all `00`).

Maybe the commands used for turning on the machine (EP3221) are different / the ones you're using have a different meaning.

machintrucbidule, 2024-02-24:
> You _could_ enable logging via the uart ports. It would interfere with communication on the bus. Using a different set of pins (for one uart) would be required. This may be possible using [...]
My issue there will be the location of the coffee machine, which is 10m away from my computer 🤣

> Enabling uart logging slows the ESP down a little, as there is more to do. Therefore timings may be different depending on the configuration. Disabling uart logging may improve/change things.
Yes i have the feeling that it induces some delay which change the behaviour

> The custom component should try to perfom the power-trip multiple times (should be visible in the logs) which is similar to turning on the power-switch multiple times. Based on your logs it looks [...]
No, no reaction at all.
But normally, when power trip is successfully completed ([D][philips_power_switch:089]: Performed 1 power trip(s).), mainboard should beep and be turned on, no ?

> What's also interesting is that the mainboard is always sending `D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26`. If I recall correctly this messages tells the display to turn off? The third byte (in this case `01`) is usually `00` for normal status update messages. After turning on some LEDs are usually lit, in this case they are all intended to be off (all `00`).
> 
> Maybe the commands used for turning on the machine (EP3221) are different / the ones you're using have a different meaning.
> 
> Sidenote: In the configuration above your still using this the link of this repository. Depending on how you compile your binaries you may want to change this to your fork/a local copy of the repository.
Yes, I already forked it, and tried to change some configuration values (such as MAX_POWER_TRIP_COUNT)
But here I'm a little bit lost, and don't know what to try :/

machintrucbidule, 2024-02-24:
In fact with only logging on the mainboard (and not on display), I have again the behaviour with mainboard turning on, and display not the first time, then turning on if I click again on the power button (and in a few cases, everything works on the first click)

Logs from mainboard only, after the first click on power button, ending with mainboard on, display off (buttons not reacting):

> [13:17:03][D][switch:012]: 'Power' Turning ON.
> [13:17:03][D][uart_debug:114]: >>> D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00
> [13:17:03][D][uart_debug:114]: >>> 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10
> [13:17:03][D][uart_debug:114]: >>> D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00
> [13:17:03][D][uart_debug:114]: >>> 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25
> [13:17:03][W][component:214]: Component api took a long time for an operation (0.11 s).
> [13:17:03][W][component:215]: Components should block for at most 20-30ms.
> [13:17:03][D][uart_debug:114]: >>> D5 55 01 01 03 00 0E 00 00 00 01 25 D5 55 01 01 03 00 0E 00 00 00 01 25
> [13:17:04][W][component:214]: Component esphome.coroutine took a long time for an operation (1.00 s).
> [13:17:04][W][component:215]: Components should block for at most 20-30ms.
> [13:17:04][D][uart_debug:114]: <<< D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00
> [13:17:04][D][uart_debug:114]: >>> 00
> [13:17:04][D][philips_power_switch:089]: Performed 1 power trip(s).
> [13:17:04][D][switch:055]: 'Power': Sending state ON
> [13:17:04][D][uart_debug:114]: <<< 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00
> [13:17:04][D][uart_debug:114]: >>> D5 55 00 01 04 02 04 00 00 00 2F 0D D5 55 00 01 04 02 04 00 00 00 2F 0D
> [13:17:04][D][uart_debug:114]: <<< 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00
> [13:17:04][D][uart_debug:114]: >>> D5 55 00 01 04 02 04 00 00 00 2F 0D D5 55 00 01 04 02 04 00 00 00 2F 0D D5 55 00 01 04 02 04 00
> [13:17:04][D][uart_debug:114]: <<< 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00
> [13:17:04][D][uart_debug:114]: >>> 00 00 2F 0D
> [13:17:04][D][uart_debug:114]: <<< 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26 D5 55 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 26 D5 55 00 03 00 07 00 00 00 00 00 00 00 00 00 00 00 03 2C D5 55 00 03 00 07 00 00 00 00 00
> [13:17:04][D][uart_debug:114]: <<< 00 00 00 00 00 00 03 2C D5 55 00 03 00 07 00 00 00 00 00 00 00 00 00 00 00 03 2C
> [13:17:05][D][switch:055]: 'Power': Sending state OFF

And logs when I click again on the power button, which end by everything on and working:

> [13:18:54][D][switch:012]: 'Power' Turning ON.
> [13:18:54][D][uart_debug:114]: >>> D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00
> [13:18:54][D][uart_debug:114]: >>> 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00 0E 00 00 00 2A 10 D5 55 0A 01 03 00  [... cut ...]
> [13:18:57][D][uart_debug:114]: <<< D5 55 00 07 07 07 3F 38 00 00 00 00 07 00 00 00 00 34 1C
> [13:18:57][D][uart_debug:114]: >>> 04 00 00 00 2F 0D D5 55 00 01 04 02 04 00 00 00 2F 0D D5 55 00 01 04 02 04 00 00 00 2F 0D D5 55
> [13:18:57][D][uart_debug:114]: <<< D5 55 00 07 07 07 3F 38 00 00 00 00 07 00 00 00 00 34 1C D5 55 00 07 07 07 3F 38 00 00 00 00 07
> [13:18:57][D][uart_debug:114]: >>> 00 01 04 02 04 00 00 00 2F 0D D5 55 00 01 04 02 04 00 00 00 2F 0D D5 55 00 01 04 02 04 00 00 00
> [13:18:57][D][text_sensor:064]: 'Status': Sending state 'Idle'

TillFleisch, 2024-02-24:
I'm also a little confused by this. Can use [this](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/issues/7#issuecomment-1941624536) debug configuration to retreive messages from the display when turning it on the machine manually? Thus we can double check, if the messagses used in the EP3221 are the same as in the EP3243.

TillFleisch, 2024-02-24:
`D5 55 0A 01 04 02 04 00 00 00 30 29` - pre-power on?
`D5 55 01 01 04 02 04 00 00 00 1B 1C` - power on? (don't know if this is with / without cleaning)

Note that when using this way of debugging, the three bytes at the end are added. Two of them (`00`) are padding the last one encodes how often the message has been seen.

It looks like these messages are different from both the `EP2220` and `EP3243`. If these starting messages are different, then the other ones are likely also different. They can be retrieved in a similar fashion by pressing the each button and observing the logs.

With these newly acquired messages you may be able to turn on the machine by using them in the component.

machintrucbidule, 2024-02-24:
Not sure the other ones are different, as during my first test I've been able to select different kind of coffees, size, etc...

And if based on EP220 and EP3243 configuration that third byte is the real message (0A for pre power, 02 for power with cleaning and 01 withour cleaning), I guess the message was a poser without cleaning.

So I tried to update the commands.h file in my fork, to update these 3 messages : https://github.com/machintrucbidule/ESPHome-Philips-Smart-Coffee/blob/series-3200/components/philips_coffee_machine/commands.h
```
        const std::vector<uint8_t> command_pre_power_on =
            {0xD5, 0x55, 0x0A, 0x01, 0x04, 0x02, 0x04, 0x00, 0x00, 0x00, 0x2A, 0x10};
        const std::vector<uint8_t> command_power_with_cleaning =
            {0xD5, 0x55, 0x02, 0x01, 0x04, 0x02, 0x04, 0x00, 0x00, 0x00, 0x1C, 0x17};
        const std::vector<uint8_t> command_power_without_cleaning =
            {0xD5, 0x55, 0x01, 0x01, 0x04, 0x02, 0x04, 0x00, 0x00, 0x00, 0x01, 0x25};
```

But nothing turns on

> 
> [14:33:33][D][switch:012]: 'Power' Turning ON.
> [14:33:34][W][component:214]: Component esphome.coroutine took a long time for an operation (0.75 s).
> [14:33:34][W][component:215]: Components should block for at most 20-30ms.
> [14:33:34][D][uart_debug:114]: >>> D5 55 0A 01 04 02 04 00 00 00 2A 10 00 00 06
> [14:33:34][D][philips_power_switch:089]: Performed 1 power trip(s).
> [14:33:34][D][switch:055]: 'Power': Sending state ON
> [14:33:34][D][number:012]: 'Coffee beans': Sending state nan
> [14:33:34][D][number:012]: 'Coffee size': Sending state nan
> [14:33:34][D][uart_debug:114]: >>> D5 55 01 01 04 02 04 00 00 00 01 25 00 00 06
> [14:33:35][D][switch:055]: 'Power': Sending state OFF

Here I don't understand what's triggering the last line "'Power': Sending state OFF" ?

machintrucbidule, 2024-02-24:
I forgot to update the 2 last bits, which I just done (at least for pre power on and power on without cleaning), still no starting, but now I see the command "D5 55 0A 01 04 02 04 00 00 00 30 29" which is "power off" sent just after the pre power. It could explain the "Sending state OFF", but any idea why?

> [14:51:53][W][component:214]: Component esphome.coroutine took a long time for an operation (0.75 s).
> [14:51:53][W][component:215]: Components should block for at most 20-30ms.
> [14:51:53][D][uart_debug:114]: >>> D5 55 00 01 04 02 04 00 00 00 2F 0D 00 00 06
> [14:51:53][D][uart_debug:114]: >>> D5 55 0A 01 04 02 04 00 00 00 30 29 00 00 06
> [14:51:53][D][philips_power_switch:089]: Performed 1 power trip(s).
> [14:51:53][D][switch:055]: 'Power': Sending state ON
> [14:51:53][D][uart_debug:114]: >>> D5 55 01 01 04 02 04 00 00 00 1B 1C 00 00 06
> [14:51:54][D][switch:055]: 'Power': Sending state OFF

TillFleisch, 2024-02-26:
> [14:51:54][D][switch:055]: 'Power': Sending state OFF

This state is published, once the display stops sending updates. This is the case on the EP2220 when the display is turned off.

> now I see the command "D5 55 00 01 04 02 04 00 00 00 2F 0D" which is "power off" sent just before (or after, it depends) the pre power

Maybe, the mainboard doesn't like being turned on the way this component currently handles power on (or the messages are invalid). On the EP3221 the power-on sequence could be different.

You can replace this component with [ssiebs MITM](https://github.com/ssieb/esphome_components/tree/master/components%2Fuart_mitm) component to observe the communication on the bus without any interference from this component. (When using the machine manually, this component should'nt mess with any messages on the bus, so it *should* make no difference if you use this or the MITM component for logging)

machintrucbidule, 2024-02-26:
Thanks, I'll check that when I'll have some time.
And yesterday I tried to replace the ESP32 S2 Mini by an ESP8266 D1 mini, I still have the same behaviour, but now the device seems to stay online (it was due to a wifi/network issue I guess)