TillFleisch/ESPHome-Philips-Smart-Coffee #37: Tests with philips ep 2339/40

finipini, 2024-02-09:
Hi, 
 I have an philips EP 2339/40 ( LatteGo series) that i would like to try this component on it , and talk about it here so someone else with the same machine could help and get some help to.  I already read some "issues" here and get some ideas of what i could do. I am a litlle busy this week but i plan to maybe make a fork off this component , as you sugested to other users.

To be honest o already tried somethings. I made a litlle "pcb" with a perfored board using an esp32 wroom:

for the uart_mainboard i used tx_pin: GPIO1, rx_pin: GPIO3 and for the uart_display i used tx_pin: GPIO17 ,rx_pin: GPIO16.  This is already hardware uarts that the esp32 have so i guess i,m doing the right think with it .

For the power_pin i first used  GPIO12, but i had the problem about it beeing an strapping pin, so the esp didnt boot with all connected, so i tried one thing, connect first to the main board, wait for the esp32 to boot and then conect to the display. With this i could control somethings in the machine ( used the yaml example) make cofee, select expresso and start/pause. And when i selected "cappucino" the state of the machine says " steam selected"  so i guess the code is similar to my machine. I couldn,t thow turn off the machine, and select the coffee size and water size in the machine it self , and the number entities in HA didnt work either.

But first i switch the power_pin to the GPIO13, and now the esp32 can boot with everythink connected ,but what happens is that the machine turns on with all the 4 leds on ( expresso, coffee, capuccino, hot water) and the aqua clean led orange(litting), when it should be blue after the clean cycle. And the machine doesnt respond to the buttons not even in HA.

I,m using an mosfet IRLZ44N as sugested , but maybe mine it,s too big has i bouht it to control led strips.  I think the next test i will do is take off the mosfet and try the other commands, to see if this is a problem of the mosfet.

TillFleisch, 2024-02-09:
>  i selected "cappucino" the state of the machine says " steam selected" so i guess the code is similar to my machine

That may be the case. Both the `EP2220` and `EP2235` use the same/almost the same protocol. From what I found online the `EP2339` is like the `EP2235` just a little different. I would guess they use the same protocol.
The current implementation offers an option to change the cappuccino/steam name. After #35 has been merged this option will be replaced with model selection.

> I,m using an mosfet IRLZ44N as sugested

If the esp manages to turn on the display it's unlikely that it's a mosfet/transistor/power-trip related issue.

I would recommend double checking you wiring. To reduce complexity you can replace this component with [ssiebs MITM](https://github.com/ssieb/esphome_components/tree/master/components%2Fuart_mitm) component.
When using the MITM component (or this component) there should be no difference to the unmodified version of coffee machine. Both components just forwarded everything from one bus to the other.

finipini, 2024-02-09:
When using the MITM component (or this component) there should be no difference to the unmodified version of coffee machine. Both components just forwarded everything from one bus to the other.

finipini, 2024-02-09:
Oh ok, that was one of my doubts, so everything should work normally in machine buttons is everything os well connected. Good to know.

finipini, 2024-02-10:
Update:

I tested the board outside the machine and saw that now gpio13 don,t do anything , even if disconected from the mosfet. i switched to gpio14 and it,s working outside the machine , but its not working connected to the machine. I measure the board outside the machine (with the mosfet) and when i turn on in home assistant the gpio goes on-off-on-off 1 second each state ( i guess it have to do with power_trip_delay).

Now in the code i put the gpio13 has the power_pin , althow nothing its connected to it ; and put the gpio14 ( connected to the mosfet) has a simple switch, and now turning the power on and the switch 14 at the same time the machine goes on and stays on with everything working.

Don,t know why this happens but has i am here with the machine working i will first try to put all the functions working ( has i see you already make the options for capuccino and the select otions for expresso etc....)  and then get back to the power issue.

finipini, 2024-02-12:
Soon after my last update i tested with the component `github://TillFleisch/ESPHome-Philips-Smart-Coffee@increase_max_repetitions_fix_power_on`       and everything worked. I can turn on the machine by esphome/HA.

I tested all the other functions and all seem to work great too. Didnt test every function but almost all.

But today i came across a wierd bug that i haven,t realise yesterday.....  If i turn on the machine by HA and i choose hot water or cappuccino ( by ha or the buttons in the machine)   the hot water nor the steam for the milk of the cappucino cames out, i hear the steam and hot water being prepared but it doesnt came out althow the hot water goes for the dust tray, maybe a safe release valve being actuated.  I guess that some byte is missing to open the valve for the hot water/steam.

If i turn on the machine by it,s buttons, and select hot water in HA or in the buttons it all works.

So i thing it is missing some message in the turn on command.   

Tonight i will try to debug what mesage my machine sends on turn on and maybe fork this project to make some tests.

TillFleisch, 2024-02-13:
> Soon after my last update i tested with the component github://TillFleisch/ESPHome-Philips-Smart-Coffee@increase_max_repetitions_fix_power_on and everything worked.

Just as a side note: This fix has already been merged into main. See #34.

> maybe a safe release valve being actuated. I guess that some byte is missing to open the valve for the hot water/steam.

I doubt that this is the case. From what we've learned so far about the protocol the display only sends messages containing the button states (i.e. which button is pressed), and not instructions on how to operate the machine (this is what the mainbooard side is responsible for).
It may be the case that the  commands from the ep2220 are interpreted incorrectly, but this depends on the specific Mainboard implementation.

Note that when preparing hot water/steam the coffee outlet nozzle is NOT used. Instead steam/hot water leaves via the steam wand. This is intended behaviour as using the coffee nozzle requires water passing through the brewing group/chamber.

finipini, 2024-02-13:
> Just as a side note: This fix has already been merged into main. See #34.
> 

Yes i saw that after i was searching for the branch i was using in your repository and didnt find it. I then reused "main" and everything continues to work.

> I doubt that this is the case. From what we've learned so far about the protocol the display only sends messages containing the button states (i.e. which button is pressed), and not instructions on how to operate the machine (this is what the mainbooard side is responsible for). 
> 

Well i tested sometimes and it really seems is like that. With your codes ( turn_on , with or without cleaning cicle)  i never get the steam or hot water out, althow, has i said , i can ear it is being prepare . And i tested various ways ( turn_on( with or without cleaning cycle) by esphome then use the hw buttons;  turn_on by esphome then use the esphome buttons;  turn_on by hw buttons then use the esphome buttons; etc.... )

I was debuging my machine codes and get a litlle diferent codes for starting :

  Your codes:

`        const std::vector<uint8_t> command_pre_power_on =
            {0xD5, 0x55, 0x0A, 0x01, 0x02, 0x00, 0x02, 0x00, 0x00, 0x00, 0x0E, 0x12};
        const std::vector<uint8_t> command_power_with_cleaning =
            {0xD5, 0x55, 0x02, 0x01, 0x02, 0x00, 0x02, 0x00, 0x00, 0x00, 0x38, 0x15};
        const std::vector<uint8_t> command_power_without_cleaning =
            {0xD5, 0x55, 0x01, 0x01, 0x02, 0x00, 0x02, 0x00, 0x00, 0x00, 0x25, 0x27};`

And what i changed:

`        const std::vector<uint8_t> command_pre_power_on =
            {0xD5, 0x55, 0x0A, 0x00, 0x01, 0x01, 0x15, 0x00, 0x00, 0x00, 0x18, 0x37};
        const std::vector<uint8_t> command_power_with_cleaning =
            {0xD5, 0x55, 0x01, 0x00, 0x01, 0x01, 0x15, 0x00, 0x00, 0x00, 0x33, 0x02};
        const std::vector<uint8_t> command_power_without_cleaning =
            {0xD5, 0x55, 0x00, 0x00, 0x01, 0x01, 0x15, 0x00, 0x00, 0x00, 0x07, 0x13};`

I made a fork and am using it now and that "bug" dissapear... well sort of . With turn_on with cleaning cycle the steam and hot water came out , with turn_on without cleaning cycle the steam and hot water dont came out either.

And i see this is not new, someone else have spotted this:  [see this forks, second last pointed issue](https://github.com/thiesgerken/esphome-philips-ep2231/tree/main?tab=readme-ov-file#ideas-for-improvement)