TillFleisch/ESPHome-Philips-Smart-Coffee #60: Suddenly lots of issues with previously working EP3243

quack3d, 2024-04-05:
I'm not sure how it could've happened but today when I turned on the coffee machine it stopped behaving normally. State said 'Preparing' like normal, but then suddenly 'Brewing Latte Macchiato', back to 'Preparing', back to 'Brewing Latte Machiato' and back to 'Preparing'. It did that 4-5 times before it said Idle. And now all lights are on and there's a blinking aqua clean. Buttons on display works sometimes and sometimes not depending if reinsert power to the wall etc. It's also not possible to turn it off anymore. Doing it from HA makes it turn on immediately afterwards. Very inconsistent behaviour. Could it be an update to ESPHome? Other ideas?

TillFleisch, 2024-04-05:
I am currently running ESPHome 2024.3.1 without issues. There have also not been any major changes recently to this repository. According to the manual `blinking aqua clean` means that the AquaClean filter needs to be replaced, maybe this is related. In your situation I would try the following steps for troubleshooting (in no particular oder/combination):
- remove other components which may interfere with this component (due to long delay/reset)
- Rollback to an older version of ESPHome
- Rollback to an older version of this repository
- use the [MITM component](https://github.com/ssieb/esphome_components) instead of this component to rule out issues with this repostiory 
- unmodify your coffee machine to see if the issue persists

> Buttons on display works sometimes and sometimes not 

To me this sound like the ESP may be restarting repeatedly for some reason. Do the logs provide any valuable information?

quack3d, 2024-04-05:
I was now able to get it back to "normal". I think I did it by reinserting power, but before it would start by itself I hit Power on with clean in HA. Then it started normally (but still with the state going from Cleaning to Brewing Latte Macchiato to Cleaning etc...). When it was turned on normally it said the water was empty. Perhaps it got in a weird limbo state due to that?

But it' still weird that Americano no longer reports Americano and that it thinks it's brewing Latte Macchiato when it's cleaning. Could this have anything to do with https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/commit/d9bb61a7b78d7898c62d5bac9912eb59d40d2083 ?

TillFleisch, 2024-04-05:
> Then it started normally

Does the "normal" behaviour persist among multiple restarts / power outages?

> Could this have anything to do with https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/commit/d9bb61a7b78d7898c62d5bac9912eb59d40d2083 ?

It should not but, you can test your hypothesis by reverting that commit/using a prior version.

quack3d, 2024-04-05:
Yeah, it seems to be "normal" through multiple restarts etc. Should this work to get the version prior to that commit?

  - source: github://TillFleisch/ESPHome-Philips-Smart-Coffee/tree/v3.0.0

It didn't make a difference ufortunately. When I turn OFF the machine I get this:

```
[18:06:57][D][text_sensor:064]: 'Status': Sending state 'Cleaning'
[18:06:58][D][text_sensor:064]: 'Status': Sending state 'Brewing Latte Macchiato'
[18:06:59][D][text_sensor:064]: 'Status': Sending state 'Cleaning'
[18:07:01][D][text_sensor:064]: 'Status': Sending state 'Brewing Latte Macchiato'
[18:07:02][D][text_sensor:064]: 'Status': Sending state 'Cleaning'
[18:07:04][D][text_sensor:064]: 'Status': Sending state 'Brewing Latte Macchiato'
[18:07:05][D][text_sensor:064]: 'Status': Sending state 'Cleaning'
[18:07:07][D][text_sensor:064]: 'Status': Sending state 'Brewing Latte Macchiato'
[18:07:08][D][text_sensor:064]: 'Status': Sending state 'Cleaning'
[18:07:10][D][text_sensor:064]: 'Status': Sending state 'Brewing Latte Macchiato'
[18:07:11][D][text_sensor:064]: 'Status': Sending state 'Cleaning'
[18:07:13][D][switch:055]: 'Power with clean': Sending state OFF
```

quack3d, 2024-04-05:
Sorry, I misunderstood the documentation. I now did this:

  - source: github://TillFleisch/ESPHome-Philips-Smart-Coffee@v3.0.0

And the Americano button shows Americano selected and turning off/on doesn't mention 'Brewing Latte Macchiato'. 

So something after 3.0.0 is not behaving well with my machine. 😊

TillFleisch, 2024-04-06:
Looks like [d9bb61a](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/commit/d9bb61a7b78d7898c62d5bac9912eb59d40d2083) might be the cause here. In particular [L177](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/blob/main/components%2Fphilips_coffee_machine%2Ftext_sensor%2Fstatus_sensor.cpp#L117).

```c++
(data[6] == led_on || data[6] == led_third)
```
This will likely fix the issue. I can double check and commit (next week).

PR #63 (2024-04-17): Fix EP32xx state detection
This PR fixes an issue on EP32xx machines which was introduced by #45.

Try this PR using:

```yaml
external_components:
  - source: github://TillFleisch/ESPHome-Philips-Smart-Coffee@fix_EP3243_state_detection
    refresh: 30min
```

Close #60
closed 2024-04-17 by TillFleisch