TillFleisch/ESPHome-Philips-Smart-Coffee #66: 'Water empty' shown when there is an actual error

bkbartk, 2024-04-18:
When all 3 warning lights are blinking (including the water tank empty one)
the status shows 'Water Empty'
but all 3 blinking at the same time means error.

I still don't know exactly what error I have, I need to powercycle to fix this. 
not sure if I can grep any debug information to provide some help.

(the error is not caused by the integration as I have seen this one a couple of times in the week before I installed this one)

TillFleisch, 2024-04-18:
I wasn't aware there was a 3 blinking leds error state as I have not come across it. The current 'error' state refers to the warning sing led. (I.e. missing coffee ground container).

Is there a way to reproduce the 3 blinking warning lights state manually?

According to the manual this 'internal error' is related to an improperly installed aqua clean filter/overheating/clogged machine.

Any recommendations for a better name than 'Internal Error'?

bkbartk, 2024-04-19:
I don't know how to manually reproduce this error.
when it happens powercycling the machine helps.

unfortunately I do not know the real cause of the error. internet search learned me that in most cases the grinder is clogged or broken. but also when the lights start to blink can help indecating the cause,
in my case this is on turn on,
but for others it's while start brewing.

so unless the machine send more information then the 3 blinking lights I think 'Internal Error' would be the only thing we know.

If there is some way to send the raw signal from the machine to a sensor in HA I can check the state of the sensor the next time the issue occurs.

bkbartk, 2024-04-25:
I forked and made a few small changes to determine the internal error
https://github.com/bkbartk/ESPHome-Philips-Smart-Coffee/blob/main/components/philips_coffee_machine/text_sensor/status_sensor.cpp (lines 70 till 83)

I can't test until the issue occurs again, that's why I haven't made a pull request yet.

I also tried to find if there was any more information available in the hex value.
but I couldn't get this value to convert to string. and even if I could I wouldn't know which byte to select.
after reading protocal.md I gave up because I don't think there is room for extra information in the protocol.

bkbartk, 2024-04-25:
you are right about the translation issue, My first goal was to make it work in whatever language, but I added the translations.
I was guessing led_second would mean blinking, but if byte 15 is used for 2 led's there are more possible outcomes
In that case `data[15] != led_off ` probably would be better.
and maybe byte 15 holds even more information.

So I tried converting data[15] to string, but ended up converting the complete array to a string according to this example
https://stackoverflow.com/questions/57399227/convert-c-byte-array-to-a-c-string
my c++ knowledge is limited so for methods and functions I just need to google

bkbartk, 2024-05-03:
I created and tested this one for internal error message
https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/pull/68
I tried to add more information but failed to do so.
(German and Italian translations are done by google translate as my German is not good enough )

I added
UART debug logging and found something interesting
```
10:30:12	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13
10:30:14	[D]	[uart_debug:114]	>>> 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36
10:30:14	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:14	[D]	[uart_debug:114]	>>> D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00
10:30:15	[D]	[uart_debug:114]	>>> D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00
10:30:15	[D]	[uart_debug:114]	>>> 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36
10:30:15	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:15	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:16	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:16	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:16	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:17	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:18	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:18	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:18	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:19	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:20	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13 D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13
10:30:21	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13 D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13
10:30:23	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13
10:30:23	[D]	[uart_debug:114]	>>> 02 00 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36
10:30:23	[D]	[uart_debug:114]	>>> D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36
10:30:23	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13 D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13
10:30:23	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13
10:30:23	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13 D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13
10:30:23	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13 D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13
10:30:23	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:23	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:24	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:24	[D]	[uart_debug:114]	>>> D5 55 00 01 02 00 20 00 00 11 36 D5 55 01 02 00 02 00 00 00 11 36 5D 15 90 90 10 A0 F3
10:30:26	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:26	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:26	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:26	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13
10:30:27	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13
10:30:27	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13
10:30:28	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13
10:30:28	[D]	[uart_debug:114]	>>> 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01
10:30:28	[D]	[uart_debug:114]	>>> D5 55 00 01 02 00 20 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00 00
10:30:28	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:28	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:29	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:29	[D]	[uart_debug:114]	>>> 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36
10:30:31	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:31	[D]	[uart_debug:114]	>>> D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36 15 90 90 10 50 F3
[... cut ...]
10:30:37	[D]	[uart_debug:114]	>>> 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01 02 00 02 00 00 00 11 36 D5 55 00 01
10:30:37	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 39 0D
10:30:37	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13
10:30:37	[D]	[uart_debug:114]	<<< D5 55 00 00 00 00 00 00 00 00 00 00 00 00 38 3F 00 12 13
```
I don't know if this is everything after I needed to open de webinterface after the lights started blinking.
but in some scenario the message from the machine holds more then 19 characters.

PR #68 (2024-06-08): Internal Errror
add internal error message

Close #66
closed 2024-06-08 by TillFleisch