TillFleisch/ESPHome-Philips-Smart-Coffee #69: EP32xx state detection still buggy

quack3d, 2024-05-29:
Opening a new issue instead of commenting on the closed  https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/pull/63. 

I'm still having issues where turning on the machine jumps between the states 'Brewing Latte Macchiato' and 'Cleaning'/'Preparing'. Below is log in HA of turning on the machine, waiting till it's idle, pressing coffee and start and then about 30 seconds after it's brewed I press the off button. Notice all the Lattes supposedly being brewed.

My graphs in HA therefore show that I brew about 50 fake Latte Macchiatos every day. Is there something I can do to make it easier to find the issue?

Or could a workaround be to add a small delay before updating state after machine has been turned on?

```ESP32 Kaffemaskin Status endret til Off
17:43:47 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Cleaning
17:43:42 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Brewing Latte Macchiato
17:43:42 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Cleaning
17:43:33 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Brewing Latte Macchiato
17:43:33 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Cleaning
17:43:31 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Idle
17:43:02 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Brewing Coffee
17:42:01 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Coffee selected
17:41:57 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Idle
17:35:35 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Cleaning
17:35:19 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Brewing Latte Macchiato
17:35:18 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Cleaning
17:35:10 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Preparing
17:35:07 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Brewing Latte Macchiato
17:35:06 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Preparing
17:35:04 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Brewing Latte Macchiato
17:35:03 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Preparing
17:35:01 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Brewing Latte Macchiato
17:35:00 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Preparing
17:34:58 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Brewing Latte Macchiato
17:34:57 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Preparing
17:34:52 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Brewing Latte Macchiato
17:34:51 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Preparing
17:34:46 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Brewing Latte Macchiato
17:34:46 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Preparing
17:34:43 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Brewing Latte Macchiato
17:34:42 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Preparing
17:34:40 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Brewing Latte Macchiato
17:34:39 - For 4 timer siden
ESP32 Kaffemaskin Status endret til Preparing
17:34:38 - For 4 timer siden```

quack3d, 2024-06-06:
I've currently changed #define REPEAT_REQUIREMENT from 30 to 100. Powering on the machine does no longer mention any Lattes. I will test brewing tomorrow.

> [22:15:43][D][switch:012]: 'Power with clean' Turning ON.
[22:15:43][W][component:237]: Component esphome.coroutine took a long time for an operation (499 ms).
[22:15:43][W][component:238]: Components should block for at most 30 ms.
[22:15:44][D][philips_power_switch:089]: Performed 1 power trip(s).
[22:15:44][D][switch:055]: 'Power with clean': Sending state ON
[22:15:44][D][switch:055]: 'Power without clean': Sending state ON
[22:15:48][D][text_sensor:064]: 'Status': Sending state 'Preparing'
[22:16:00][D][text_sensor:064]: 'Status': Sending state 'Cleaning'
[22:16:24][D][text_sensor:064]: 'Status': Sending state 'Idle'

quack3d, 2024-06-07:
Done some more testing now. REPEAT_REQUIREMENT to 40 showed only sporadic events of Lattes when starting the machine (very seldom). I'm currently using it at 50 and it seems solid. Is there any disadvantage in increasing this to 50? I'm not fully understanding what it does. 

If 50 works for everyone else I'm suggesting to change it to that. Otherwise perhaps a configurable setting?

TillFleisch, 2024-06-08:
The `REPEAT_REQUIREMENT` was added for this exact purpose, to remove intermittent states.
The state estimation isn't great since determining the true state of the machine requires observing the blinking state of the leds.
The current implementation has lots of room for improvement. Ideally, a state machine would be implemented which mirrors the coffee machines states (this should also consider the users input, which can be observed from the bus). The current quasi-stateless design is only observing the current state of the leds (and if the play/pause on is blinking) ,but not recent menu selections or user input.

The downside of increasing the `REPEAT_REQUIREMENT` is that the status sensor responds slower to changes.
I will increase this value shortly.

commit b36d7ce9 (2024-06-08): Increase status repeat requirement to remove intermittent states; Close #69
closed 2024-06-08 by TillFleisch