TillFleisch/ESPHome-Philips-Smart-Coffee #74: Espresso out of nowhere

dkwolf1, 2024-07-19:
Hello! I followed this project a while and now I've installed everything. Pcb and picoflex cables, I al worked perfect. But, when the machine is running idle, after let's say 15 minutes he makes an espresso out of nowhere...more people with this problem? I don't have any log files yet, I will try to get them. 

[image]

[image]

TillFleisch, 2024-07-19:
Logs would be great (before/after the event)!
Please also post your yaml config (please redact secrets).

ESPHome has a feature which resets the ESP every 15 minutes when not connected to WiFi.
A reset should not result in spontaneous espresso. It doesn't on my machine, but maybe your machine/configuration is different. You can try this by using the [restart button](https://esphome.io/components/button/restart.html).

This may also be related to #26

dkwolf1, 2024-07-19:
[logs_philips-ep_logs.txt](https://github.com/user-attachments/files/16314634/logs_philips-ep_logs.txt)
[philips-ep.txt](https://github.com/user-attachments/files/16314639/philips-ep.txt)

Now its turning off without touching a button...Searching to log the uart but can't find it yet

dkwolf1, 2024-07-20:
I think you can close this probleem. I detected a shortage on the circuit, according to the smd voltage regulator. There was 7.5v in the circuit. Now it looks all good, no problems yet

dkwolf1, 2024-07-21:
This is my config with debugging. I also measured everything. I have 3.3v on the tx0 and 5v on the rx0. When i dont connect the display everything works fine. When i connect the display something is ticking inside the coffeemachine. If i remove the debug options and connect the display it works, thicking is over. Measurements are normal.

esphome:
  name: koffie
  friendly_name: PhilipsEP

esp8266:
  board: esp01_1m

logger:
  baud_rate: 0

external_components:
  - source: github://TillFleisch/ESPHome-Philips-Smart-Coffee@main

api:
  encryption:
    key: "DGxVd69ihlyy8lUwMIrWGOnLkq2o9ZoUmiFN7APW0Mw="

ota:
  - platform: esphome
    password: "

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip: 
    static_ip: 192.168.1.245
    subnet: 255.255.255.0
    gateway: 192.168.1.1

  
  ap:
    ssid: "Koffie Fallback Hotspot"
    password: 

captive_portal:

time:
  - platform: homeassistant
    id: time_homeassistant
    on_time_sync:
      then:
        - component.update: philips_uptime_timestamp

uart:

  - id: uart_mainboard
    tx_pin: GPIO1
    rx_pin: GPIO3
    baud_rate: 115200
    debug:
      after:
        bytes: 12

  - id: uart_display
    tx_pin: GPIO15
    rx_pin: GPIO13
    baud_rate: 115200
    debug:
      after:
        bytes: 12

philips_coffee_machine:
  id: philip
  display_uart: uart_display
  mainboard_uart: uart_mainboard
  power_pin: GPIO12
  power_message_repetitions: 5
[... 189 lines cut ...]
[21:27:17][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[21:27:17][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[21:27:17][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[21:27:17][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[21:27:17][C][wifi_info:011]: WifiInfo SSID 'Netwerk'
[21:27:17][C][wifi_info:011]:   Icon: 'mdi:access-point-network'
[21:27:17][C][wifi_info:009]: WifiInfo IPAddress 'IP adres'
[21:27:17][C][wifi_info:009]:   Icon: 'mdi:ip-network'
[21:27:41][D][switch:055]: 'Power': Sending state ON
[21:27:41][D][switch:055]: 'Power zonder schoonmaak': Sending state ON
[21:27:41][D][number:012]: 'Koffiebonen': Sending state nan
[21:27:41][D][number:012]: 'Koffie grootte': Sending state nan
[21:27:42][D][switch:055]: 'Power': Sending state OFF
[21:27:42][D][switch:055]: 'Power zonder schoonmaak': Sending state OFF
[21:27:43][D][sensor:093]: 'philips_uptime': Sending state 44.79500 s with 0 decimals of accuracy
[21:27:51][D][switch:055]: 'Power': Sending state ON
[21:27:51][D][switch:055]: 'Power zonder schoonmaak': Sending state ON
[21:27:53][D][text_sensor:064]: 'Status': Sending state 'Preparing'
[21:27:59][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[21:28:20][D][text_sensor:064]: 'Status': Sending state 'Cleaning'
[21:28:30][D][sensor:093]: 'PhilipsEP wifi signaal': Sending state -58.00000 dBm with 0 decimals of accuracy
[21:28:43][D][sensor:093]: 'philips_uptime': Sending state 104.79500 s with 0 decimals of accuracy
[21:28:45][D][text_sensor:064]: 'Status': Sending state 'Idle'
[21:29:43][D][sensor:093]: 'philips_uptime': Sending state 164.79700 s with 0 decimals of accuracy
[21:30:30][D][sensor:093]: 'PhilipsEP wifi signaal': Sending state -56.00000 dBm with 0 decimals of accuracy

TillFleisch, 2024-07-21:
You only need the debug option on one of the UART (preferably the mainboard one), as they bot the the same messages (aside from those which are injected by this component, hence the mainbaord preference).
This cuts the amount of messages which are sent in half.

IIRC, I have at some point had an ESP which made a clicking noise when the LED was blinking (fast) or during high WiFi loads.

Can you describe the clicking noise further? Is it more of a buzzing sound, how loud is it? Can you locate where it's coming from? (the custom PCB?)
My **guess** would be that the debug messages increase the power draw of the ESP as it's busy using the radio, and this somehow affects the rest of the system.  Does the ticking noise disappear when you don't observe logs?

[Here is a debug version](https://github.com/TillFleisch/ESPHome-Philips-Smart-Coffee/issues/7#issuecomment-1941624536)  which can cut down on messages even more, but it's only for messages coming from the display and it may require modification based on the machine that is being used.

Based on the altered/removed message: are surprise espressos / shutdowns still a thing? Also make sure that the machine is still working as intended without the modification to rule out that something broke.

dkwolf1, 2024-07-22:
The clicking noise is from inside the machine. When i change the debug its al fine. I think something else is going on. This night there was nothing wrong, no turning on automaticly, no ticking and no espressos. This morning the automation worked fine, get my coffee when i woke up. During an hour no strange behavior. But then, there was an update from Esphome. 2024.7.1. After this update, say 10 minutes, the device give me an espresso. I reset the device but no results. Measuements on PCB are normal. 

[09:16:29][D][uart_debug:114]: >>> 01:03:00:0D:00:00:00:02:12:D5:55:00
[09:16:29][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:29][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:29][D][uart_debug:114]: >>> 01:03:00:0D:00:00:00:02:12:D5:55:00
[09:16:29][D][uart_debug:114]: >>> 01:03:00:0D:00:00:00:02:12:D5:55:00
[09:16:29][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:29][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:29][D][uart_debug:114]: >>> 01:03:80:43:00:00:02:12:D5:55:00:01
[09:16:29][D][uart_debug:114]: >>> 00:02:12:D5:55:00:01:03:00:0D:00:00
[09:16:29][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:29][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:29][D][uart_debug:114]: >>> 00:02:12:D5:55:00:01:03:00:0D:00:00
[09:16:29][D][uart_debug:114]: >>> 00:02:12:D5:55:00:01:03:00:0D:00:00
[09:16:29][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:29][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:29][D][uart_debug:114]: >>> 00:02:12:D5:55:00:01:03:00:0D:00:00
[09:16:29][D][uart_debug:114]: >>> 00:02:12:D5:55:00:01:03:00:0D:00:00
[09:16:29][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:29][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:29][D][uart_debug:114]: >>> 00:02:12:D5:55:00:01:03:00:0D:40:00
[09:16:29][D][uart_debug:114]: >>> 02:12:D5:55:00:01:03:00:0D:00:00:00
[09:16:29][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:29][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:29][D][uart_debug:114]: >>> 02:89:D5:55:00:01:03:00:0D:00:00:00
[09:16:29][D][uart_debug:114]: >>> 02:12:D5:55:00:01:03:00:0D:00:00:00
[09:16:29][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:29][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:29][D][uart_debug:114]: >>> 02:12:D5:55:00:01:03:00:0D:00:00:00
[09:16:29][D][uart_debug:114]: >>> 02:12:D5:55:00:40:20:0D:00:00:00:02
[09:16:29][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:29][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:29][D][uart_debug:114]: >>> 12:75:55:D0:00:0D:00:00:00:02:12:D5
[09:16:29][D][uart_debug:114]: >>> 55:00:01:03:00:0D:00:00:00:02:12:D5
[09:16:29][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:29][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:30][D][uart_debug:114]: >>> 55:00:01:03:00:0D:00:00:00:02:12:D5
[09:16:30][D][uart_debug:114]: >>> 55:00:01:03:00:0D:00:00:00:02:12:D5
[09:16:30][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:30][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:30][D][uart_debug:114]: >>> 0D:00:00:00:02:12:D5:55:00:01:03:00
[09:16:30][D][uart_debug:114]: >>> 0D:00:00:00:02:12
[09:16:30][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:30][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:30][D][uart_debug:114]: >>> D5:55:00:01:03:00:0D:00:00:00:02:12
[09:16:30][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:30][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:30][D][uart_debug:114]: >>> D5:55:00:01:03:00:0D:00:00:00:02:12
[09:16:30][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:30][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:30][D][uart_debug:114]: >>> D5:55:00:01:03:00:0D:00:00:00:02:12
[09:16:30][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:30][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:30][D][uart_debug:114]: >>> D5:55:00:01:03:00:0D:00:00:00:02:12
[09:16:30][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:30][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:30][D][uart_debug:114]: >>> D5:55:00:01:03:00:0D:00:00:00:02:12
[09:16:30][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:30][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:30][D][uart_debug:114]: >>> D5:55:00:01:03:00:0D:00:00:00:02:12
[09:16:30][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:30][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:30][D][uart_debug:114]: >>> D5:55:00:01:03:00:0D:00:00:00:02:12
[09:16:30][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:30][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:30][D][uart_debug:114]: >>> 03:00:0D:00:00:00:02:12:D5:55:00:01
[09:16:30][D][uart_debug:114]: >>> 03:00:0D:00:00:00:02:12:D5:55:00:01
[09:16:30][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:30][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[... 65 lines cut ...]
[09:16:31][D][uart_debug:114]: >>> D5:55:00:01:03:00:0D:00:00:00:02:12
[09:16:31][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:31][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:31][D][uart_debug:114]: >>> D5:55:00:01:03:00:0D:80:00:02:12:D5
[09:16:31][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:31][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[... cut ...]
[09:16:31][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:31][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:31][D][uart_debug:114]: >>> D5:55:00:01:03:00:0D:00:00:00:02:12
[09:16:31][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00
[09:16:31][D][uart_debug:114]: <<< 07:00:00:00:00:32:32
[09:16:31][D][uart_debug:114]: >>> D5:55:00:01:03:00:0D:00:00:00:02:12
[09:16:31][D][uart_debug:114]: <<< D5:55:00:07:07:07:07:00:00:00:00:00

Sending this, keeps going on and on.

TillFleisch, 2024-07-22:
Hm, ... this is wired behavior. 
Based on the config file, I presume you are performing the automation through HA and not on the ESP itself (using ESPHome). Is that correct?
I would recommend the following trouble shooting steps:
* remove any automation and see if the issue persists
* remove all coffee machine related buttons (potentially also other components) from the config and see if the issue persists
  * add back features until the issue re-appears

If you remove the espresso buttons from the config, does the erratic behavior happen with other beverages?
Does the machine actually brew coffee or is it only the status sensor which is reporting this behavior?

dkwolf1, 2024-07-25:
Today I do it al over again. 

1. I measure the board: voltage and ohm: nothing weird.
2. I measure the cables: all good

So no hardware problem. Now i go over to the software, starting like you said. Post the results later this day

closed 2024-09-09 by TillFleisch

xi-s, 2024-12-23:
I replaced my ESP32 (it stopped working) by a ESP8266 board. Since the change, I'm also getting espressos out of nowhere!

Still need to investigate further, but apparently dkwolf1 is not the only one who has experienced this problem.

@dkwolf1, did you solve it already?

dkwolf1, 2024-12-23:
@xi-s no...i even quit the project. But what you say now is opening my eyes again. I always worked with the 8266...should is be the 32 then...

xi-s, 2024-12-23:
My first try to solve the problem is downgrading to the exact same version I was running on my ESP32

(ESPHome 2024.4.0 + This project @ V3.0.3)

Results pending!

xi-s, 2024-12-27:
I moved from using GPIO3|GPIO1 + GPIO13|GPIO15 to GPIO3|GPIO1 + GPIO14|GPIO12.

So far I'm on 18 hours of "idle" state without random espresso

TillFleisch, 2024-12-29:
> I moved from using GPIO3|GPIO1 + GPIO13|GPIO15 to GPIO3|GPIO1 + GPIO14|GPIO12.
> So far I'm on 18 hours of "idle" state without random espresso

Interesting....

Any chance you have serial logging still enabled? Make sure to use `baud_rate: 0` in you esphome config as this could interfere with communication on the bus(that particular UART).

xi-s, 2024-12-29:
I made sure to include the `baud_rate: 0`

Current working config; The only difference between random espresso and normal is the choice of GPIO right now:

```
esphome:
  name: Coffee
  friendly_name: Philips EP2230

# NodeMCU ESP8266
esp8266:
  board: nodemcuv2

# Enable logging
logger:
  #Disable serial logging
  baud_rate: 0

api:

ota:
  platform: esphome

wifi:
  ssid: <redacted>
  password: <redacted>
  use_address: <redacted>

external_components:
  - source: github://TillFleisch/ESPHome-Philips-Smart-Coffee@main

uart:
  # UART connected to the mainboard
  - tx_pin: GPIO12
    rx_pin: GPIO14
    baud_rate: 115200
    id: uart_mainboard
  # UART connected to the display
  - tx_pin: GPIO1
    rx_pin: GPIO3
    baud_rate: 115200
    id: uart_display

philips_coffee_machine:
  display_uart: uart_display
  mainboard_uart: uart_mainboard
  power_pin: GPIO16
  id: philips
  model: EP_2235

text_sensor:
  - platform: philips_coffee_machine
    controller_id: philips
    id: status
    name: "Status"

switch:
  - platform: philips_coffee_machine
    controller_id: philips
    name: "Power"
    icon: mdi:coffee

button:
  - platform: philips_coffee_machine
    controller_id: philips
    action: SELECT_ESPRESSO
    name: "Espresso"
    icon: mdi:coffee
  - platform: philips_coffee_machine
    controller_id: philips
    action: SELECT_COFFEE
    name: "Coffee"
    icon: mdi:coffee
  - platform: philips_coffee_machine
    controller_id: philips
    action: SELECT_HOT_WATER
    name: "Hot water"
    icon: mdi:water
  - platform: philips_coffee_machine
    controller_id: philips
    action: SELECT_CAPPUCCINO
    name: "Cappuccino"
    icon: mdi:coffee
  - platform: philips_coffee_machine
    controller_id: philips
    action: PLAY_PAUSE
    name: "Play/pause"
    icon: mdi:play-pause
number:
  - platform: philips_coffee_machine
    type: bean
    name: "Strength"
    controller_id: philips
    status_sensor_id: status
  - platform: philips_coffee_machine
    type: size
    name: "Size"
    controller_id: philips
    status_sensor_id: status
```