TillFleisch/ESPHome-Philips-Smart-Coffee #75: EP3241 Power Off Cleaning Cycle

joelguth, 2024-07-23:
Hey there!

First off awesome work on this project!

I just finished my PCB assembly and am testing in an EP3241, which seems to be working relatively well with the EP3243 commands. One thing I've noticed is that if I turn off the machine with the "switch" from esphome or from the display, it shuts the machine down and runs a clean cycle when it turns off. If I just let the machine sit idle, it seems the machine just immediately turns off as if power is cut. 

Is there a difference with how the idle timeout is handled in the program loop vs that of turning the machine off? 

Also, is there a way to extend the time before this idle shutdown happens? It seems to happen fairly quickly if no buttons are pressed on the machine.

I could also be missing something very simple configuration wise. Happy to collect any additional debug information if necessary.

Thank you!

joelguth, 2024-07-23:
Here's some debug logs. It just randomly immediately shuts down when i'm selecting different beverages and sizes from the esphome device:

```
[15:38:40][D][uart_debug:114]: >>> D5 55 00 01 03 00 12 00 00 00 10 21 00 00 0E
[15:38:40][D][uart_debug:114]: >>> D5 55 00 01 03 00 12 00 00 00 10 21 00 00 09
[15:38:40][D][uart_debug:114]: >>> D5 55 00 01 03 00 12 00 00 00 10 21 00 00 0A
[15:38:42][D][uart_debug:114]: >>> D5 55 00 01 03 00 12 00 00 00 10 21 00 00 63
[15:38:43][D][button:010]: 'Select Latte' Pressed.
[15:38:43][D][uart_debug:114]: >>> D5 55 00 01 03 00 12 00 00 00 10 21 00 00 49
[15:38:43][D][number:012]: 'Beverage Strength': Sending state 2.000000
[15:38:43][D][number:012]: 'Beverage Size': Sending state 2.000000
[15:38:43][D][uart_debug:114]: >>> D5 55 00 01 03 00 0E 10 00 00 2D 24 00 00 06
[15:38:44][D][text_sensor:064]: 'Status': Sending state 'Latte Macchiato selected'
[15:38:44][D][number:012]: 'Milk Size': Sending state 2.000000
[15:38:48][D][uart_debug:114]: >>> D5 55 00 01 03 00 12 00 00 00 10 21 00 00 F2
[15:38:48][D][number:054]: 'Milk Size' - Setting number value
[15:38:48][D][number:113]:   New number value: 3.000000
[15:38:48][D][uart_debug:114]: >>> D5 55 00 01 03 00 12 00 00 00 10 21 00 00 19
[15:38:48][D][uart_debug:114]: >>> D5 55 00 01 03 00 0E 00 08 00 1F 16 00 00 06
[15:38:48][D][number:012]: 'Milk Size': Sending state 3.000000
[15:38:49][D][uart_debug:114]: >>> D5 55 00 01 03 00 12 00 00 00 10 21 00 00 31
[15:38:49][D][uart_debug:114]: >>> D5 55 00 01 03 00 12 00 00 00 10 21 00 00 20
[15:38:50][D][number:054]: 'Beverage Size' - Setting number value
[15:38:50][D][number:113]:   New number value: 3.000000
[15:38:50][D][uart_debug:114]: >>> D5 55 00 01 03 00 12 00 00 00 10 21 00 00 43
[15:38:50][D][number:012]: 'Beverage Size': Sending state 3.000000
[15:38:51][D][uart_debug:114]: >>> D5 55 00 01 03 00 0E 00 04 00 04 07 00 00 05
[15:38:51][D][number:054]: 'Beverage Strength' - Setting number value
[15:38:51][D][number:113]:   New number value: 3.000000
[15:38:51][D][uart_debug:114]: >>> D5 55 00 01 03 00 12 00 00 00 10 21 00 00 24
[15:38:51][D][uart_debug:114]: >>> D5 55 00 01 03 00 0E 00 02 00 2D 2D 00 00 06
[15:38:51][D][number:012]: 'Beverage Strength': Sending state 3.000000
[15:38:52][D][number:012]: 'Beverage Strength': Sending state nan
[15:38:52][D][number:012]: 'Beverage Size': Sending state nan
[15:38:52][D][number:012]: 'Milk Size': Sending state nan
[15:38:52][D][switch:055]: 'Power': Sending state OFF
[15:38:52][D][text_sensor:064]: 'Status': Sending state 'Off'
```

Here's a snippet of my esphome config:

```
philips_coffee_machine:
  display_uart: uart_display
  mainboard_uart: uart_mainboard
  power_pin: GPIO12
  id: philip
  model: EP_3243

text_sensor:
  - platform: philips_coffee_machine
    controller_id: philip
    id: status
    name: "Status"

switch:
  - platform: philips_coffee_machine
    controller_id: philip
    name: "Power"
    clean: false
    icon: mdi:coffee-maker

button:
  - platform: philips_coffee_machine
    controller_id: philip
    action: SELECT_COFFEE
    name: "Select Coffee"
    icon: mdi:coffee
    
  - platform: philips_coffee_machine
    controller_id: philip
    action: SELECT_LATTE
    name: "Select Latte"
    icon: mdi:coffee

  - platform: philips_coffee_machine
    controller_id: philip
    action: SELECT_AMERICANO
    name: "Select Americano"
    icon: mdi:coffee

  - platform: philips_coffee_machine
    controller_id: philip
    action: SELECT_CAPPUCCINO
    name: "Select Cappuccino"
    icon: mdi:coffee

  - platform: philips_coffee_machine
    controller_id: philip
    action: SELECT_ESPRESSO
    name: "Select Espresso" 
    icon: mdi:coffee

  - platform: philips_coffee_machine
    controller_id: philip
    action: SELECT_HOT_WATER
    name: "Select Hot Water"
    icon: mdi:tea

  - platform: philips_coffee_machine
    controller_id: philip
    action: PLAY_PAUSE
    name: "Select play/pause"
    icon: mdi:play-pause

number:
  - platform: philips_coffee_machine
    type: bean
    name: "Beverage Strength"
    controller_id: philip
    status_sensor_id: status

  - platform: philips_coffee_machine
    type: size
    name: "Beverage Size"
    controller_id: philip
    status_sensor_id: status

  - platform: philips_coffee_machine
    type: milk
    name: "Milk Size"
    controller_id: philip
    status_sensor_id: status
```

TillFleisch, 2024-07-24:
> Is there a difference with how the idle timeout is handled in the program loop vs that of turning the machine off?

The software switch does the exact same as pressing the button on the display. If some type of coffe has been prepared, the machine will perform a cleaning cycle during shutdown.
The ESPHome switch does not turn the machine off automatically, unless explicitly specified through some automation/config.
Note, that the coffee machines have an internal timeout for shutting down, which can be changed IIRC (see manual).

Does the same behaviour appear if you remove the power switch from you configuration?

Also make sure you don't have any automations running that could interfere with the normal operation of the device.

A log of the Mainboard TX UART would be great , maybe this contains some useful information. The provided log looks 'normal' at first glance.

The EP3241 could also be misinterpreting the commands for all I know... Depends on the machines implementation. Is this behaviour reproducible with a specific type of beverage?

There is already an open discussion regarding the EP3241: see #55
We should move this discussion there if this isn't an issue with the existing implementation.
You could also try to use the commands from the related thread to see if they resolve the issue.

TillFleisch, 2024-08-13:
Let us continue any related discussion in #55 until we have working EP3241 support

closed 2024-08-13 by TillFleisch

fmarkf, 2024-10-20:
Hi,

Not sure is this is a good idea to comment in the closed topic.
However I have exactly the same issue as the OP.  My machine is EP3241.

All the buttons and statuses seems to work fine (including physical controls), however the machine is turning OFF randomly. 
I can clearly see in the logs that there is an Off command being sent. I don't have any automation running.

I tried removing Power switch from the configuration - still random machine Off commands. 
Any ideas how to diagnose/debug this issue?

P.S.: One other issue - the machine stopped remembering last selected 'beverage size, strength' settings. 

[11:40:07][C][api:141]:   Address: 192.168.1.20:6053
[11:40:07][C][api:143]:   Using noise encryption: YES
[11:40:11][D][switch:012]: 'P3200 Power' Turning ON.
[11:40:11][W][component:237]: Component esphome.coroutine took a long time for an operation (501 ms).
[11:40:11][W][component:238]: Components should block for at most 30 ms.
[11:40:11][D][philips_power_switch:089]: Performed 1 power trip(s).
[11:40:11][D][switch:055]: 'P3200 Power': Sending state ON
[11:40:14][D][text_sensor:064]: 'P3200 Status': Sending state 'Preparing'
[11:40:30][D][text_sensor:064]: 'P3200 Status': Sending state 'Cleaning'
[11:40:35][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[11:40:56][D][text_sensor:064]: 'P3200 Status': Sending state 'Idle'
[11:41:05][D][switch:055]: 'P3200 Power': Sending state OFF
[11:41:05][D][text_sensor:064]: 'P3200 Status': Sending state 'Off'

fmarkf, 2024-10-25:
In case anyone comes across similar issue with EP3241... Switching from Wemos D1 Mini (ESP8266) to ESP32 DevKit board resolved random Power Off commands for me. Go figure... :)

TillFleisch, 2024-11-12:
Good to know that there is a patch to this problem, thank you for providing updates. Seems like the underlying problem may be ESP8266 specific code. Let's continue this discussion in #55.