TillFleisch/ESPHome-Philips-Smart-Coffee #83: Random behavior - not brewing

p1ranha, 2024-09-26:
Hi, I finally managed to receive your original assembled PCB from JLC today, and after mounting the connectors, I programmed the ESP.

The whole machine started behaving erratically, I am not even able to control it properly using the screen nor ESP commands. I have following config:
philips_coffee_machine:
  display_uart: uart_display
  mainboard_uart: uart_mainboard
  power_pin: GPIO12
  id: coffee_machine
  language: en-US
  model: EP_3246

It seems like reset cycle comes in after the grinding and when brewing starts. Nothing comes out to the glass.

After manually turning on the machine, it waits for selection. I select espresso and press play.
It start grinding, espresso is lit and then after grinding, all the buttons get animation of the circle. No coffee actually came out.

In the log, I see following:
14:35:56 | [D] | [switch:055] | 'Power': Sending state ON
-- | -- | -- | --
14:36:00 | [D] | [text_sensor:064] | 'Status': Sending state 'Preparing'
14:36:05 | [D] | [text_sensor:064] | 'Status': Sending state 'Idle'
14:36:08 | [D] | [sensor:093] | 'Service Interval Counter': Sending state 8.00000 count with 0 decimals of accuracy
14:36:11 | [D] | [sensor:093] | 'Brew Count': Sending state 8.00000 count with 0 decimals of accuracy
14:36:14 | [D] | [text_sensor:064] | 'Status': Sending state 'Espresso selected'
14:36:35 | [D] | [text_sensor:064] | 'Status': Sending state 'Brewing Espresso'
14:36:53 | [D] | [text_sensor:064] | 'Status': Sending state 'Preparing'
14:37:02 | [D] | [text_sensor:064] | 'Status': Sending state 'Idle'
Since no espresso came out, I tried coffee by selecting coffee and pressing play:
14:37:06 | [D] | [text_sensor:064] | 'Status': Sending state 'Coffee selected'
14:37:08 | [D] | [sensor:093] | 'Service Interval Counter': Sending state 9.00000 count with 0 decimals of accuracy
14:37:11 | [D] | [sensor:093] | 'Brew Count': Sending state 9.00000 count with 0 decimals of accuracy
14:37:15 | [D] | [text_sensor:064] | 'Status': Sending state 'Preparing'
14:37:28 | [D] | [text_sensor:064] | 'Status': Sending state 'Idle'
14:38:08 | [D] | [sensor:093] | 'Service Interval Counter': Sending state 9.00000 count with 0 decimals of accuracy
14:38:11 | [D] | [sensor:093] | 'Brew Count': Sending state 9.00000 count with 0 decimals of accuracy

There was no coffee coming out, and after grinding the animated loop of all buttons went thru.

Then without pressing nothing, I see some button presses happening:
**14:38:35 | [D] | [text_sensor:064] | 'Status': Sending state 'Hot water selected'** | nothing was pressed
14:38:38 | [D] | [text_sensor:064] | 'Status': Sending state 'Cleaning'
14:38:45 | [D] | [text_sensor:064] | 'Status': Sending state 'Preparing'
14:38:53 | [D] | [switch:055] | 'Power': Sending state OFF
14:38:53 | [D] | [switch:055] | 'PowerWithClean': Sending state OFF
14:38:53 | [D] | [text_sensor:064] | 'Status': Sending state 'Off'

Any ideas what may be wrong?

p1ranha, 2024-09-26:
This is when I just press "select espresso" and wait in the ESPhome:

It marks as espresso selected, then after some time it starts brewing it, never actually outputs anything and then turns itself off.

14:51:57 | [D] | [switch:012] | 'Power' Turning ON.
-- | -- | -- | --
14:51:58 | [W] | [component:237] | Component esphome.coroutine took a long time for an operation (500 ms).
14:51:58 | [W] | [component:238] | Components should block for at most 30 ms.
14:51:58 | [D] | [philips_power_switch:089] | Performed 1 power trip(s).
14:51:58 | [D] | [switch:055] | 'Power': Sending state ON
14:51:58 | [D] | [switch:055] | 'PowerWithClean': Sending state ON
14:51:59 | [D] | [text_sensor:064] | 'Status': Sending state 'Preparing'
14:52:08 | [D] | [sensor:093] | 'Service Interval Counter': Sending state 9.00000 count with 0 decimals of accuracy
14:52:10 | [D] | [text_sensor:064] | 'Status': Sending state 'Idle'
14:52:10 | [D] | [button:010] | 'Select Espresso' Pressed.
14:52:11 | [D] | [sensor:093] | 'Brew Count': Sending state 9.00000 count with 0 decimals of accuracy
14:52:12 | [D] | [text_sensor:064] | 'Status': Sending state 'Espresso selected'
14:52:18 | [D] | [text_sensor:064] | 'Status': Sending state 'Brewing Espresso'
14:52:26 | [D] | [text_sensor:064] | 'Status': Sending state 'Preparing'
14:52:40 | [D] | [text_sensor:064] | 'Status': Sending state 'Idle'
14:52:47 | [D] | [text_sensor:064] | 'Status': Sending state 'Hot water selected'
14:52:51 | [D] | [text_sensor:064] | 'Status': Sending state 'Cleaning'
14:53:00 | [D] | [text_sensor:064] | 'Status': Sending state 'Preparing'
14:53:06 | [D] | [switch:055] | 'Power': Sending state OFF
14:53:07 | [D] | [switch:055] | 'PowerWithClean': Sending state OFF
14:53:07 | [D] | [text_sensor:064] | 'Status': Sending state 'Off'

p1ranha, 2024-09-26:
and the yaml config:

```
substitutions:
  name: "philips-coffee-machine"
  friendly_name: "Coffee Machine"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  name_add_mac_suffix: true

  project:
    name: TillFleisch.ESPHome-Philips-Smart-Coffee
    version: "1.0"

esp8266:
  board: esp12e

logger:
  baud_rate: 0

api:

ota:
  platform: esphome
web_server:
  port: 80

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.3.192
    gateway: 192.168.3.1
    subnet: 255.255.255.0
  ap:
    ssid: "Philips Fallback Hotspot"
    password: !secret wifi_password

time:
  - platform: homeassistant
    id: homeassistant_time

globals:
  - id: service_interval_counter
    type: int
    restore_value: true
    #initial_value: '0'
  - id: brew_count
    type: int
    restore_value: true
    #initial_value: '0'  

captive_portal:

external_components:
  - source: github://TillFleisch/ESPHome-Philips-Smart-Coffee@main

uart:
  # UART connected to the mainboard
  - tx_pin: GPIO1
    rx_pin: GPIO3
    baud_rate: 115200
    id: uart_mainboard

  # UART connected to the display
  - tx_pin: GPIO15
    rx_pin: GPIO13
    baud_rate: 115200
    id: uart_display

philips_coffee_machine:
  display_uart: uart_display
  mainboard_uart: uart_mainboard
  power_pin: GPIO12
  id: coffee_machine
  language: en-US
  model: EP_3246

text_sensor:
  - platform: philips_coffee_machine
    controller_id: coffee_machine
    id: status
    name: "Status"
    on_value:
      then:
        - lambda: |-
            if (id(status).state.find("Brewing 2x") != std::string::npos) {
              id(brew_count) += 2;
              id(service_interval_counter) += 2;
            } else if (id(status).state.find("Brewing") != std::string::npos) {
              id(brew_count) += 1;
              id(service_interval_counter) += 1;
            }       
switch:
  - platform: philips_coffee_machine
    controller_id: coffee_machine
    name: "Power"
    clean: false
    icon: mdi:coffee-maker

  - platform: philips_coffee_machine
    controller_id: coffee_machine
    name: "PowerWithClean"
    icon: mdi:coffee-maker

button:
  - platform: restart
    name: $friendly_name Restart

  - platform: philips_coffee_machine
    controller_id: coffee_machine
    action: SELECT_COFFEE
    name: "Select Coffee"
    icon: mdi:coffee
  - platform: philips_coffee_machine
    controller_id: coffee_machine
    action: MAKE_COFFEE
    name: "Make Coffee"
    icon: mdi:coffee

  - platform: philips_coffee_machine
    controller_id: coffee_machine
    action: SELECT_ESPRESSO
    name: "Select Espresso" 
    icon: mdi:coffee
  - platform: philips_coffee_machine
    controller_id: coffee_machine
    action: MAKE_ESPRESSO
    name: "Make Espresso" 
    icon: mdi:coffee

  - platform: philips_coffee_machine
    controller_id: coffee_machine
    action: PLAY_PAUSE
    name: "Select play/pause"
    icon: mdi:play-pause

sensor:
  - platform: template
    name: "Service Interval Counter"
    lambda: |-
      return id(service_interval_counter);
    state_class: measurement
    unit_of_measurement: "count"
    accuracy_decimals: 0

  - platform: template
    name: "Brew Count"
    lambda: |-
      return id(brew_count);
    state_class: measurement
    unit_of_measurement: "count"
    accuracy_decimals: 0     
```

p1ranha, 2024-09-27:
I also just checked, and it never even created a single coffee puck. I just see grinded coffee in the coffee waste bin. 
so it somehow resets between grinding and making a puck. I disconnected the board today, and the machine works fine.

quack3d, 2024-09-27:
Can you show your wiring? I don't use a PCB but I had lots of erratic behaviour when I used an ESP8266. When I switched to ESP32 it worked fine.

p1ranha, 2024-09-27:
I ordered the original PCB with all parts assembled, so there is not much special wiring i have done.

[image]

[image]

TillFleisch, 2024-09-27:
Which coffee machine model do you have?

> I disconnected the board today, and the machine works fine.

That's good, the PCB also looks good.

Unfortunately, I have not yet heard of this kind of behaviour and since this modification does not change the inner working of the machine it's unexpected to see this kind of mischievous behaviour.

This component assumes that the `EP_3246` is the same as the `EP_3243` as the commands have been proven to work on the `EP_3246` before. It *may* be the case that your machine is different for some reason.

I would recommend starting with a minimal configuration to figure out which part causes problems.
Use a minimal configuration (generated by ESPHome) and add the external component as well the UART config and the `philips_coffee_machine` component.
```yaml
philips_coffee_machine:
  display_uart: uart_display
  mainboard_uart: uart_mainboard
  power_pin: GPIO12
  id: coffee_machine
  language: en-US
  model: EP_3246
```

This *should* not affect the machine, as the component is only forwarding messages in both directions. Please test if the erroneous behaviour persists (while the machine is idling/has a beverage selected). If this does not help, the following configuartion can simplify the setup even more.
Use a minimal ESPHome configuration, add the UART config from above and use [this UART MITM external component](https://github.com/ssieb/esphome_components/tree/master/components%2Fuart_mitm).

Please also make sure nothing external is affecting your setup (an automation in Home Assistant for instance).
I would also like to point out that ESPHome restarts the ESP every 15 min. if it's not connected to HA or WiFi, which may cause problems.

This could also be a power-related issue. Does the machine beep in the above scenario?
Does the LED on the ESP blink when the machine resets/aborts?

p1ranha, 2024-09-30:
Sorry, I was busy this weekend, and can only answer now.
The model is: Philips 3200 series ‎EP3246/70 bought in May 2022
There are no automations yet, and no issues connecting to wifi. The led doesn't blink.

I have started with minimal yaml:
```
esphome:
  name: "philips-coffee-machine"
  friendly_name: "Coffee Machine"

esp8266:
  board: esp01_1m

# Enable logging
logger:
  baud_rate: 0
# Enable Home Assistant API
api:
  encryption:
    key: *redacted*

ota:
  - platform: esphome
    password: *redacted*

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.3.192
    gateway: 192.168.3.1
    subnet: 255.255.255.0
  ap:
    ssid: "Philips Fallback Hotspot"
    password: !secret wifi_password

captive_portal:

external_components:
  - source: github://TillFleisch/ESPHome-Philips-Smart-Coffee@main

globals:
  - id: service_interval_counter
    type: int
    restore_value: true
    #initial_value: '0'
  - id: brew_count
    type: int
    restore_value: true
    #initial_value: '0' 

uart:
  # UART connected to the mainboard
  - tx_pin: GPIO1
    rx_pin: GPIO3
    baud_rate: 115200
    id: uart_mainboard

  # UART connected to the display
  - tx_pin: GPIO15
    rx_pin: GPIO13
    baud_rate: 115200
    id: uart_display

philips_coffee_machine:
  display_uart: uart_display
  mainboard_uart: uart_mainboard
  power_pin: GPIO12
  id: coffee_machine
  language: en-US
[... 63 lines cut ...]
[12:35:31][C][mdns:116]: mDNS:
[12:35:31][C][mdns:117]:   Hostname: philips-coffee-machine
[12:35:31][C][esphome.ota:073]: Over-The-Air updates:
[12:35:31][C][esphome.ota:074]:   Address: 192.168.3.192:8266
[12:35:31][C][esphome.ota:075]:   Version: 2
[12:35:31][C][esphome.ota:078]:   Password configured
[12:35:31][C][safe_mode:018]: Safe Mode:
[12:35:31][C][safe_mode:019]:   Boot considered successful after 60 seconds
[12:35:31][C][safe_mode:021]:   Invoke after 10 boot attempts
[12:35:31][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[12:35:31][C][api:139]: API Server:
[12:35:31][C][api:140]:   Address: 192.168.3.192:6053
[12:35:31][C][api:142]:   Using noise encryption: YES
[12:36:18][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[12:36:49][D][text_sensor:064]: 'Status': Sending state 'Preparing'
[12:37:24][D][text_sensor:064]: 'Status': Sending state 'Idle'
[12:37:31][D][text_sensor:064]: 'Status': Sending state 'Brewing Espresso'
[12:37:40][D][text_sensor:064]: 'Status': Sending state 'Preparing'
[12:37:54][D][text_sensor:064]: 'Status': Sending state 'Idle'
[12:38:02][D][text_sensor:064]: 'Status': Sending state 'Espresso selected'
[12:38:05][D][text_sensor:064]: 'Status': Sending state 'Brewing Espresso'
[12:38:16][D][text_sensor:064]: 'Status': Sending state 'Preparing'
[12:38:28][D][text_sensor:064]: 'Status': Sending state 'Idle'

Below you can find the whole video what happens since I press Power button till I press Espresso. It basically starts brewing on it's own, and resets itself abruptly. The led doesn't flash. (I cut down the part where it's heating up and nothing happens).

p1ranha, 2024-10-07:
Do you see anything else I could try to get it resolved?

bkbartk, 2024-11-25:
If I'm correct, this component only forwards messages from the display to the machine and the other way around.
also some codes are injected, but nothing should be changed.
So in this case is should only forward espresso, and handle the response.

maybe something is not assembled correctly,
how many PCB's did you order from JLC ?
I thought the minimum was, so maybe you can try another one or replace the picoflex cable you added.

TillFleisch, 2024-12-02:
Sorry for the late response!

> If I'm correct, this component only forwards messages from the display to the machine and the other way around.

This component forwards traffic in both directions, otherwise the display and mainboard would not be able to talk to each other in both directions.
Commands are only injected towards the mainboard side (spoofing of display functionality).

To rule out that this is a HW issue (which I don't think it is), you can try using [ssiebs MITM](https://github.com/ssieb/esphome_components/tree/master/components%2Fuart_mitm) component instead of this external component. If the issue goes away, it's likely a bug in this software.

p1ranha, 2024-12-02:
Unfortunately I had to give up on that as troubleshooting exceeded my skills.

closed 2024-12-02 by p1ranha

bkbartk, 2024-12-02:
I think the only thing you have to do is create a yaml like this
```
esphome:
  name: "philips-coffee-machine"
  friendly_name: "Coffee Machine"

esp8266:
  board: esp01_1m

# Enable logging
logger:
  baud_rate: 0
# Enable Home Assistant API
api:
  encryption:
    key: *redacted*

ota:
  - platform: esphome
    password: *redacted*

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.3.192
    gateway: 192.168.3.1
    subnet: 255.255.255.0
  ap:
    ssid: "Philips Fallback Hotspot"
    password: !secret wifi_password

captive_portal:

external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome_components
    components: [ uart_mitm ]

uart:
  # UART connected to the mainboard
  - tx_pin: GPIO1
    rx_pin: GPIO3
    baud_rate: 115200
    id: uart_mainboard

  # UART connected to the display
  - tx_pin: GPIO15
    rx_pin: GPIO13
    baud_rate: 115200
    id: uart_display

uart_mitm:
  uart1: uart_mainboard
  uart2: uart_display
```
upload it to the device and check if it works,
It not it's probably a HW issue,
if so, it's a component issue.

- one remark, I did not test this one.

CamelCase1072, 2025-03-15:
Hey @p1ranha, 
I don't know if this helps to resolve your issue but I experienced the same issue on the same modell using the alternative PCB from thommy86 and a Wemos D1 Mini (with the ESP6288). Since I had ordered 5 of the PCBs I even tried a new PCB and a new D1 Mini but the issue didn't resolve. 
I replaced the D1 Mini with a D1 Mini ESP32 and since then it works well. So my assumtion is that some ESP6288 boards are not powerful enough for this application or at least have some issue with this specific model.

p1ranha, 2025-03-15:
Oh thank you so much for that hint. As of now I gave up on this project but may revisit it back with your idea.
Once more thanks for your input ♥️