TillFleisch/ESPHome-Philips-Smart-Coffee #90: EP2220/10: Power on issues

cr-insane, 2025-01-02:
Hi,
first of all, thanks for your great work! :-)

I have an EP2220/10 and I am having issues with the power on switch. I already played around with `power_message_repetition` and `power_trip_delay` but nothing really worked.
Whenever I hit the power switch once, nothing happens. When I hammer it 5-10 times in a row, then the machine might beep or turn on.

I enabled debug logging from the display to the mainboard after pressing the power button.

To me it seems, that the commands my display sends are different.
They always start with that `D5:55:0A:00:00:03:02`
I couldn't find any examples using that format, whether your project or the one from chris7topher.

But I am wondering, how it's even able to function sometimes, if the commands are different.
[logs_coffe_logs(1).txt](https://github.com/user-attachments/files/18294027/logs_coffe_logs.1.txt)

TillFleisch, 2025-01-12:
`D5:55:0A:00:00:03:02` is also new to me.
This, and the log seem to contain "unkown" commands. Maybe your machine/display is running different firmware.

> But I am wondering, how it's even able to function sometimes, if the commands are different.

This may be caused by noise or some coffee machine internal logic...

Does other fucntionality work (i.e. select and make coffee)?

cr-insane, 2025-01-12:
Yeah, the other commands do work as far as I have tried.

I suspect the issue might be on my board.

Just as a disclaimer, I am not very familiar with electronic stuff.

After pressing "Power On", the GPIO12 sends pulses of 500ms or longer of 3.3v power to the mosfet, right?

I can measure the voltage going to the MOSFET, but on the drain I can never measure any voltage. How high should the voltage be on the drain?
Or am I just confused here?

l0l67, 2025-06-23:
Hi,
i also seem to have this problem.

I can use all functions fine except the turn on command (turn off, brew, switch to water, sizes, turn off).

Turn on only works if i toggle the switch in HA two or three times..

l0l67, 2025-06-23:
@ch-rit did you ever solve this?

ch-rit, 2025-06-23:
@l0l67 unfortunately, i did not.
I tried debugging the communication protocol and maybe figure out another power on cmd, but didnt get any further :/ I tried some other commands which also didnt work.

For me it also doesnt work, when the aqua clean is not blinking

william-sy, 2025-11-25:
I have seem to have fixed this problem. Also the 2025.11 build problem. but it still needs testing before i merge it to master.
please have a look here: https://github.com/william-sy/ESPHome-Philips-Smart-Coffee (branch poweron)

The-May, 2025-12-28:
> I have seem to have fixed this problem. Also the 2025.11 build problem. but it still needs testing before i merge it to master. please have a look here: https://github.com/william-sy/ESPHome-Philips-Smart-Coffee (branch poweron)

This seem to have fixed all my other issues.
This is the first time (on first glance) that it works physically AND digitally (HA)
@william-sy  I love you so much.

stamanf, 2026-03-01:
> I have seem to have fixed this problem. Also the 2025.11 build problem. but it still needs testing before i merge it to master. please have a look here: https://github.com/william-sy/ESPHome-Philips-Smart-Coffee (branch poweron)

Hello, just saw your branch and would like to give it a try.  I'm a newbie with github.  To change the repo, do I only need to change my yaml to point to your branch? Do I need to wipe my esp32 first? Anything else I need to do?

Thanks

william-sy, 2026-03-02:
@stamanf, Copy the new espyaml i have onto your esp, clean build files and flash :)